S7NetPlus/s7netplus #272: First time wanted to write entire class and happens nothing.

Mallowan, 2020-07-07:
Hello everyone,
First time I decided to write a class.

Class itself:

 ```
 public class Test
    {
        public int Value0 { get; set; }
        public int Value1 { get; set; }
        public int Value2 { get; set; }       
    }
```

Button event:

```
 private void Load_button_Click(object sender, RoutedEventArgs e)
        {

            Test  _test = new test();
            test.Value0 = 33;
            test.Value1 = 34;
       
           _service.plc.WriteClass(test, 113, 0);
                  
        }
```
There are no any changes in DB113. There are no any exceptions in project.
What I'm doing wrong?
Any insights on this matter, please.

FalcoGoodbody, 2020-07-08:
Did you open the connection first?
Can you show the structure of the DB113?

Mallowan, 2020-07-08:
[image]

FalcoGoodbody, 2020-07-08:
I think you need convert the C#-INT into the S7-INT. Or you change the S7 into DINT.
It is described in the pdf-docu at the very end.

I think that is the problem. 
I myself haven´t yet tested to write a class, sorry.

Mallowan, 2020-07-09:
@FalcoGoodbody , 
it started working when I doubled quantity of bits on the S7 side. Int --> DInt.
Since there was no a parity between C and  S7  integers.
Making _shorts_ on the side of C came to nothing. Not many made attempts, though.
Thanks a lot! It was very valuable advice.

mycroes, 2020-09-08:
Hi @Mallowan, Can I close this issue? If not, can you provide a minimal sample that fails?

closed 2021-06-04 by mycroes