S7NetPlus/s7netplus #274: Unable to use Plc.Read(DataType,startByte,varType,varCount)

brynarn, 2020-07-09:
If I try to read from DataBlock that only contains a single 16-bit integer. The read fails.
If i add another 16-bit integer after this one it works.
The error message is "WrongNumberReceivedBytes".

It seems that the socket connection always reads 4 bytes. 
Does anyone know a fix for this?

[image]
[image]
[image]

mycroes, 2020-09-08:
Can you retest this against latest S7NetPlus (0.7.0)? It should at least provide a better exception message if it still fails.

SmackyPappelroy, 2020-09-09:
It works in 0.7.0! So this issue is resolved

mycroes, 2020-09-09:
Thanks for testing!

closed 2020-09-09 by mycroes