S7NetPlus/s7netplus #275: How to handle the PLC disconnection error?

nayanavv, 2020-07-14:
How to handle the error thrown when the PLC (ethernet port) is disconnected from the windows application. How to handle this error?  By that I meant how to not suspend the thread and keep the application running.
Also I'm getting different error for ethernet disconnection and PLC power off, how is that possible?

Plc is not getting disconnected when PLC Close function is called in the form.closed event.

>  PLC.Close()

Please help with any possible solutions

kivannc, 2020-07-14:
I have a similar problem. In some computers when the is connection is gone for a second. IsConnected stays true but it always gets read error and shows wrong values. For now I have counter for errors. If it continues for a while. I reinitialize the connection.

nayanavv, 2020-07-14:
> I have a similar problem. In some computers when the is connection is gone for a second. IsConnected stays true but it always gets read error and shows wrong values. For now I have counter for errors. If it continues for a while. I reinitialize the connection.

Hi kivannc,
Could you please explain how you did this? Reinitialize means do I have to restart the application?

kivannc, 2020-07-16:
I am just Closing and Openning Connection again. Instance.Close(); Instance.Open();

mycroes, 2021-06-06:
This might also be affected by #349.