S7NetPlus/s7netplus #276: I can read a variable but I can't write

michaelsequeira86, 2020-07-23:
I trying to write a variable in a S7-300.
The read method works.

bool example:
S7.NetSiemensS7Client.Write("DB216.DBX202.0", true);

I tried bool, int, word... but nothing is write into PLC

Protection;
[image]

How can i check the errors when i write something, when all the write methods return void??

FalcoGoodbody, 2020-07-23:
First please check if you have set the protection-level at the PLC to the lowest so that you can have full access as described on the last site in the pdf-documentation.

Second: I am using, for example, the following code write a single bit into a PLC

```cs
plc.WriteBit(DataType.DataBlock, 14, 0, 0, false);  
```

michaelsequeira86, 2020-07-23:
> 
> 
> First please check if you have set the protection-level at the PLC to the lowest so that you can have full access as described on the last site in the pdf-documentation.
> 
> Second: I am using, for example, the following code to write a single bit into a PLC
> 
> ```cs
> private static Plc plc = new Plc(CpuType.S7300, "192.168.0.64", 0, 2);
> plc.Open();
> plc.WriteBit(DataType.DataBlock, 14, 0, 0, false);  
> plc.Clode();
> ```

I'm using Step7, the only protection I know is the parameter described above. Is not the same protection present in the pdf Document

I also tried to write with :
plc.WriteBit(DataType.DataBlock.........)
but nothing happened

FalcoGoodbody, 2020-07-23:
Can you change this protection level to the radio-button "1 - key position"?
You need to upload hardware-configuration after you changed this.

michaelsequeira86, 2020-07-24:
Thanks for your answer.
 The problem was that:

[image]

I increase the number of connections.... my bad

Thanks

closed 2020-07-24 by michaelsequeira86

Paul8193, 2023-12-28:
> Thanks for your answer. The problem was that:
> 
> [image]
> 
> I increase the number of connections.... my bad
> 
> Thanks

@michaelsequeira86 
I want to know whether the snapshot of PLC Communication Settings is correct for solve problem or not?  if not, would you please tell me how to increase the number of connections.  Thank you very much.