S7NetPlus/s7netplus #283: Unnecessary fail after Read (TKPT incomplete)

jakob-ledermann, 2020-08-14:
This might be related to #250 and simliar errors.
The current Implementation assumes, that a stream.Read only returns less than requested if the message is incomplete or a EOF is encountered. This assumption does not hold. NetworkStream.Read is guaranteed to block until the first byte is available and then copies the available bytes up to the requested count into the provided buffer.
If you are in a heavy load network environment (in my case 5 Mbit/s from a different device on the same network adapter + S71200) it happens that the Read function only copies a small amount of bytes into the buffer, if you call read again with an adjusted offset and the count reduced by the already received count, you often get more bytes of the expected message.

The current implementation (especially 0.4.0) fails after the first try. I hope i can get my fix up to 0.5.0 soon, so I can submit a PR.
One of the hard parts is to make sure I don't introduce an potentially endless loop.

mycroes, 2020-08-14:
Hi @jakob-ledermann, I figured this wouldn't happen because the message size is smaller than a normal MTU, so I didn't handle it in Sally7 either. It's definitely something that's worth fixing, so unless you absolutely want to tackle this yourself I'll take a jab at it.

mycroes, 2020-08-14:
Possibly related: #262 .

scamille, 2020-08-16:
https://stackoverflow.com/a/24412022 provides an example for a Stream ReadExactly extension function, which might be all that is needed here. Just needs to replace all the Stream.Read calls with it then. And a async version should be easy to add as well.

mycroes, 2020-08-17:
Thanks @scamille . I was expecting to fix it along the same lines, didn't have time to do something with it yet. I'll try to fix this this evening.

jakob-ledermann, 2020-08-17:
My internal solution, that only needs a sorting of the repositories:
`
        /// <summary>
        /// Reads a fixed amount of bytes from the stream into the buffer
        /// </summary>
        /// <param name="stream">the Stream to read from</param>
        /// <param name="buffer">the buffer to read into</param>
        /// <param name="offset">the offset in the buffer to read into</param>
        /// <param name="count">the amount of bytes to read into the buffer</param>
        /// <param name="timeoutMs">The timeout to abort the read</param>
        /// <returns>returns the amount of read bytes</returns>
        /// <exception cref="TimeoutException">
        ///  Throws timeout exception, when the timeout elapsed, while waiting for available Data
        /// </exception>
        public static int ReadExact(this NetworkStream stream, byte[] buffer, int offset, int count, int timeoutMs)
        {
            int read = offset;
            int received = 0;
            count = Math.Min(count, buffer.Length - offset);
            do
            {
                if (stream.DataAvailable)
                {
                    received = stream.Read(buffer, read, count - read);
                    read += received;
                }
                else
                {
                    var timedOut = SpinWait.SpinUntil(() => stream.DataAvailable, 10);
                    timeoutMs -= 10;
                    received = 1;
                }
            }
            while (read < count && !(timeoutMs <= 0 || received <= 0));

            if (read < count && timeoutMs <= 0)
            {
                throw new TimeoutException($"Timeout receiving desired size. Missing {count - read} bytes");
            }

            return read;
        }

        /// <summary>
        /// Reads a fixed amount of bytes from the stream into the buffer
        /// </summary>
        /// <param name="stream">the Stream to read from</param>
        /// <param name="buffer">the buffer to read into</param>
        /// <param name="offset">the offset in the buffer to read into</param>
        /// <param name="count">the amount of bytes to read into the buffer</param>
        /// <param name="timeoutMs">The timeout to abort the read</param>
        /// <returns>returns the amount of read bytes</returns>
        /// <exception cref="TimeoutException">
        ///  Throws timeout exception, when the timeout elapsed, while waiting for available Data
        /// </exception>
        public static int ReadExact(this NetworkStream stream, byte[] buffer, int offset, int count, int timeoutMs)
        {
            int read = offset;
            int received = 0;
            count = Math.Min(count, buffer.Length - offset);
            do
            {
                if (stream.DataAvailable)
                {
                    received = stream.Read(buffer, read, count - read);
                    read += received;
                }
                else
                {
                    var timedOut = SpinWait.SpinUntil(() => stream.DataAvailable, 10);
                    timeoutMs -= 10;
                    received = 1;
                }
            }
            while (read < count && !(timeoutMs <= 0 || received <= 0));

            if (read < count && timeoutMs <= 0)
            {
                throw new TimeoutException($"Timeout receiving desired size. Missing {count - read} bytes");
            }

            return read;
        }
`

This is very along the proposed solution on stackoverflow, but handles intentionally to short TKTP messages in a way the library user can proceed. The proposed solution blocks indefinitly in the case of a malformed Message.
The async Variant looks very alike, but is not yet finished.

scamille, 2020-08-17:
Shouldn't the normal solution timeout on the socket read and write timeouts automatically, so no extra timeout should be necessary.
If a user currently does not set a timeout, it will block indefinitely as well, so I don't think there needs to be a special timeout handling here.

For the async version I would welcome having a API with cancellation tokens. I might create a PR for that eventually.

mycroes, 2020-08-17:
@jakob-ledermann: I don't like the SpinWait in the sync implementation, I fully agree with @scamille that the timeout on the stream/socket will be good enough. As for your async implementation, that's quite OK, but I wouldn't provide a 'default' cancellation token. As far as I know nothing changes with regards to timeouts, so there's no reason to provide a default CancellationToken. Also, I would prefer the CancellationToken to be the last argument, but it can perfectly be optional and default to CancellationToken.None, which would solve the order for methods with optional arguments as well. Contributions to the project are most certainly welcome, so if you want to get this into PR then I'll be happy to review, otherwise I'll create a similar implementation to what has been discussed here.

@scamille: As above, contributions are welcome, but if @jakob-ledermann wants to take this up then that's fine by me too.

Thanks to both of you for the ideas, references and code.

scamille, 2020-08-17:
The problem with CancellationToken and sockets seems to be that they do not really seem to work for sockets even if the .NET API offers them. See https://stackoverflow.com/questions/12421989/networkstream-readasync-with-a-cancellation-token-never-cancels

In my Code for connecting the plc (OpenAsync), I added cancellationToken.Register to socket.Close() as suggested in the stackoverflow thread. This seems to work and allows me to cancel a connection attempt on a unreachable PLC.

I don't want to take this over, just wanted to express my interest in the subject. Both the increased stability/streaming compliance from ReadExact and Cancellation support :-)

But we should probably focus on the ReadExact / ReadExactAsync methods.

mycroes, 2020-08-17:
Agreed, ReadExact is what we should be discussing here. I'm amazed by the CancellationToken issue, that's something that affects more of my code as well.

@jakob-ledermann, are you going to create a PR to address ReadExact or would you rather have me address it?

jakob-ledermann, 2020-08-18:
The SpinWait was originally a Thread.Sleep, but that is not available in .Net Standard 1.3.

I don't like situations, where the most convenient way to use it has some hidden endless loop or simliar behaviour.
Providing the API with an optional CancellationToken parameter, which is None is in my opinion the same as providing an default CancellationToken which never cancels.

If we rely on the readtimeout of the stream I would select a default ReadTimeout earlier than never. A custom timeout could apply a different Timeout for parts of the message than for the first part of the answer (which this implementation does not).
I am not sure how much of my time I can spend on this problem.

The CancellationToken is a cooperative cancellation mechanism, so ignoring a cancellation request is allowed. I think the only usefull position to cancel and be able to use the connection further is between sending two messages to the plc.

mycroes, 2020-09-04:
Should be fixed in 0.7.0!

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes