S7NetPlus/s7netplus #287: Problem reading from DB

lemonfunk, 2020-08-18:
Hi there,
i'm just trying to read DB variables using S7.Net+ (updated vo V0.6) from an S7-1200

The first problem i had it the known issue that the "isavailable" request seems to close the socket connection. (is there a plan to fix this?)

By avoiding "isavailable" i came a bit closer to the finish line (BTW: optimization is of, access granted)
Now i get following exception when i try to perform a read 
`var result = (bool)SPS.Read("DB2.DBX72.0");`

results in:
S7.Net.PlcException: "Received 12 bytes: '32-02-00-00-00-00-00-00-00-00-81-04', expected 19 bytes."

jakob-ledermann, 2020-08-18:
If I recall correctly the last two bytes of the message are parsed as error class and error code by wireshark. I did not yet find a list of error codes and meanings or I might have added parsing them and providing more usefull messages. Maybe wireshark is able to provide a text error message.
I had a problem where I thought the configuration was correct, but I forgot to push the hardware configuration to the PLC.

Did you check the allow "PUT/GET..." part of the access configuration?

lemonfunk, 2020-08-19:
Yes Put/Get is enabled. I'm not sure if some kind of firewall / network issue could be a reason? Siemens TIA works fine between the computer and S7. I have a KTP400-HMI connected to the SPS (if this has any influence)

FalcoGoodbody, 2020-08-19:
I have to admitt that since the last version 0.6.0 (?) I also get this message very often now on different readings from PLCs (all S7-1500). I mostly read classes or words. With the version before (0.4.0) that did not occure.

Btw: it is good to see that some people started to push the development of this great library! Thank you all for efforts

lemonfunk, 2020-08-19:
Shame on me!
BTW: 0.4 gives an array index error instead of the above

BUT: It seems it really was a problem with the PUT/GET option? It was activated, shown as active during the online connection with TIA and i also updated the code several times, but maybe the put/get status was'nt yet pushed to the SPS?

I just tried to activate the HTTP server to check if the SPS gehts this change and now it works (even if deactivated webserver now)

FalcoGoodbody, 2020-08-19:
Thank you!
So there will be an update to the library sometime today?

lemonfunk, 2020-08-19:
No in my case it seems that the problem was a configuratuion issue of the S7 and not a s7.net+ issue

closed 2020-08-19 by lemonfunk

FalcoGoodbody, 2020-08-19:
In my case the problem seems to be something other. The messages just occure with the newer version of the library.
PUT/GET-permission is active since ever...that can not be the problem

jakob-ledermann, 2020-08-19:
@lemonfunk My guess is you did transfer the software to the PLC but somehow missed to transfer the hardware to the PLC. I had a simliar error and took a few hours of searching to find this problem. In my case there were some surrounding conditions which prevented the transfer of the hardware.

@FalcoGoodbody can you open a new issue and also test with Version 0.5.0? Could you also describe the value you are trying to read in more detail?

FalcoGoodbody, 2020-08-19:
I will do, thanks