S7NetPlus/s7netplus #294: getting exceptions when reading bytes with single request

FalcoGoodbody, 2020-08-20:
Since I am using the version 0.6.0 I am getting execptions when reading values from a DB in a S7-1500.

The expection is as follows:
>S7.Net.PlcException: Received 15 bytes: '32-03-00-00-00-0A-00-02-00-01-00-00-05-01-FF', expected 19 bytes. ---> S7.Net.PlcException: Received 15 bytes: '32-03-00-00-00-0A-00-02-00-01-00-00-05-01-FF', expected 19 bytes.
   bei S7.Net.Plc.AssertReadResponse(Byte[] s7Data, Int32 dataLength)
   bei S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)

My internal log-file adresses often the same line of code as follows:

```csharp
try
 {
     if (plcKonti2.IsConnected)         <------- this line is adressed in the exception
     {
     #region Konti2ISA
     // K2 Rakel Laminierstation
     var statusRakelLaminierstation = plcKonti2.ReadBytes(DataType.DataBlock, 153, 0, 1);
     K2RakelLaminiereinheitFreigabe = statusRakelLaminierstation[0].SelectBit(0);
      }
 }
```
The connection to the plc is opened before I call the method above. The code was running for nearly 1 year without any exceptions at this line of code.
I will check if this also happens with version 0.5.0.

scamille, 2020-08-20:
This seems to be because of the stricter response integrity checks from 0b6226327b5f62138f4ee2583f690f6c89393e26 . Those are a good thing because otherwise you would not be sure that the communication was successfull.

The underlying cause is probably the RedExact problem which we have been discussing these past days, where a tcp stream can return early with less data than requested.

FalcoGoodbody, 2020-08-20:
Thank you.
I just want update the information I have got while testing version 0.5.0 and 0.4.0.

With version 0.5.0 the exception is excatly the same.
With version 0.4.0 it is an "index out of range" exception

> S7.Net.PlcException: Der Index war außerhalb des Arraybereichs. ---> System.IndexOutOfRangeException: Der Index war außerhalb des Arraybereichs.
   bei S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   --- Ende der internen Ausnahmestapelüberwachung ---
   bei S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   bei S7.Net.Plc.ReadBytes(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)

mycroes, 2020-08-20:
We're looking into fixing this. I will be on holiday for a week, I'm not sure if I'm picking up any changes or PR's in that time, but otherwise I intent to fix this soon after. These messages are really useful though because now there's actually something we can diagnose when people report issues.

mycroes, 2020-09-04:
Can you retest against 0.7.0?

FalcoGoodbody, 2020-09-07:
I compiled this morning with version 0.7.0 and started the application at 7 o´clock am. Now it is around 1 o´clock pm and the application took aroun 1.800 data logs from the different PLCs (one each 10 seconds).

scamille, 2020-09-07:
Glad to hear. Do you monitor any abnormal execution times for your test?

What I mean is assuming the communication you perform usually takes 400ms, are you able to observe outliers if it sometimes takes lets say 1s or longer?

FalcoGoodbody, 2020-09-07:
no I do not monitor any execution times at the moment. The application is logging more than 300 different values from 10 different PLCs each 10 seconds, some each 10 minutes and store it in a MSSQL database. I just checked the timestamps from the database entries and it looks like the logging is working fine. The timestamps differs only for some milliseconds.

I am "new" to C# programming, this is my first project wich is used in production. I usually do programming PLCs ;-)
I think I could implement a time monitoring, if this would help you. But I can´t say anything about the execution-times before verion 0.7.0.

scamille, 2020-09-07:
Well I would assume that when the entries in your database appear somewhat exactly every 10 seconds, then that should be good enough.
One could probably also do some SQL magic to extra the interval between each successive entry. But that's a bit above my SQL skills as well :-)

I just wanted to ask about this since you appear to have such a large real-life test laboratory set up, which is great for this project. Having Snap7 Virtual S7 PLC Server for the Unit tests is great, but it certainly does not offer the same insight as such a long running test with that amount of data.

FalcoGoodbody, 2020-09-07:
well let me see what I can do ;-) You are interessted in the time each communication need do be done? For each of the 10 plc separated and/or the overall time?
I think we can use our "large real-life test laboratory set up" here to some tests for you. So I can help a little bit instead of only using this library ;-)

scamille, 2020-09-07:
Well either the time it takes to execute the communication, or just the interval between sample.
Should definitely be grouped by PLC.

The new code in 0.7.0 does have some changes which do better TCP stream reading, but that could also mean longer execution time (instead of failure, which is good) and it would be interesting to see if there are sometimes hikkups. Of course that then also depends on the network and everything, but I would just make the bold assumption that this should work reliably ;)

FalcoGoodbody, 2020-09-08:
[Mappe1.xlsx](https://github.com/S7NetPlus/s7netplus/files/5187103/Mappe1.xlsx)
[App.log](https://github.com/S7NetPlus/s7netplus/files/5187104/App.log)

scamille, 2020-09-08:
The exceptions in your log come from the underlying .NET Socket, and they will get the error message from the operating system. So not much we can/want to do here, it just looks like your PLC or network connection for that specific PLC is sometimes down for about 10 to 20s. The first reconnection attempt does not succeed, only the second 20s after the initial read error does. After that it works again.

Interesting that you connect to the plc/plant from outside. At my workplace we usually try to place our data aquisition directly onto a server in the plant. But that is often quite a pain, so having a nice VM sounds nice. With .NET Core one could even move all of it to some docker container running on linux... Well maybe someday we are able to do that as well :P 

I looked at the data for your first PLC (Konti2) and got:
```
mean | 1030,32383076923
var | 27,2124341841329
stddev | 5,2165538609443
min | 1021,1355
max | 1046,2091
```
which looks quite decent and reliable. I assume that the amount of bytes you transfer for Konti2 is somewhat large and that is the reason the execution time is so much longer than the other ones.

Anyway, thanks for the data. It does look like this works quite reliably for now. And since you do not only read raw bytes but also seem to serialize them, that also hints at not getting any errors in the underlying hardware, otherwise you would probably get wrong values and exceptions on the more complex, non-numeric types.

mycroes, 2020-09-08:
Thanks for the info guys. Good to hear the issue is solved. It's quite interesting that you were actually running into this while I don't seem to encounter this issue with Sally7 which takes the old approach from S7NetPlus of reading once only, and we're actually using this in production to control machines (so not just reading, but a lot of writing as well) even at some 24/7 facilities. Does mean I should probably apply the same fixes there though.

I'm closing this issue because the actual issue is solved, however I absolutely endorse further discussion. Also, feel free to reopen this issue if required or to open a new issue for seperate problems/discussions.

FalcoGoodbody, 2020-09-09:
Sorry for opening this issue again but I just want to inform you, that I am getting exceptions for a while which disappear after a some time.
`2020-09-09 10:46:27.0057|INFO|S7.Net.PlcException: Received 15 bytes: '32-03-00-00-00-0A-00-02-00-01-00-00-05-01-FF', expected 19 bytes. ---> S7.Net.PlcException: Received 15 bytes: '32-03-00-00-00-0A-00-02-00-01-00-00-05-01-FF', expected 19 bytes.
   bei S7.Net.Plc.AssertReadResponse(Byte[] s7Data, Int32 dataLength)
   bei S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   --- Ende der internen Ausnahmestapelüberwachung ---
   bei S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   bei S7.Net.Plc.ReadBytes(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   bei OptiplanProzessDatenLog.MainWindow.ReadPlcKonti2() in C:\Users\Robert.Oppel.OPL-NB-008\source\repos\OptiplanProzessDatenLog\OptiplanProzessDatenLog\Form1.cs:Zeile 2018. `

Line 2018 of my code looks as follows
```csharp
try
            {                                                                     <-- this is line 2018
                if (plcKonti2.IsConnected)
                {
                    #region Konti2ISA
                    // K2 Rakel Laminierstation
                    var statusRakelLaminierstation = plcKonti2.ReadBytes(DataType.DataBlock, 153, 0, 1);
                    K2RakelLaminiereinheitFreigabe = statusRakelLaminierstation[0].SelectBit(0);
                    K2RakelLaminiereinheitAnlagenTeilEin = statusRakelLaminierstation[0].SelectBit(1);
                    K2RakelLaminiereinheitStatus = (K2RakelLaminierstationState)plcKonti2.ReadStruct(typeof(K2RakelLaminierstationState), 153, 2);
                    K2RakelLaminiereinheitPumpeBetriebsstunden = ((uint)plcKonti2.Read("DB154.DBD0")).ConvertToInt();
```

Yesterday I read in one of the closed issues from @mycroes that one should not use the "IsConnected" parameter.
Can someone confirm that?

mycroes, 2020-09-09:
@FalcoGoodbody, no I was wrong, I'm not sure if there was an issue with IsConnected before, but calling IsAvailable after connecting defintely was broken. You are using S7NetPlus 0.7.0, right?

FalcoGoodbody, 2020-09-09:
right. 
I changed my code a little bit in that way, that I now close the connection to the plc if this execption occurs. Let´s see how this works.

scamille, 2020-09-09:
Just as a note: I do read/write single-bytes in my production code as well ("handshakes" to see if the PLC has new data), and that does work just fine in general.

Your message is quite strange. You get 15 bytes, which is the core response, with 0xff at position 14 which indicated status OK.

Since you request exactly 1 byte, I do think 18+1 = 19 is the correct amount of bytes to expect, since the payload is created in ReadBytesWithSingleRequest by the following line:
```Array.Copy(s7data, 18, buffer, offset, count);```

But since you do net get a TPKT exception (which we would get when the stream returns EOF), the PLC is telling is that this truly what it thinks the payload should be. And it does look okish, it's just too short.

mycroes, 2021-06-06:
Affected by #349.

scamille, 2021-06-06:
It might also be resolved by the synchronization added in 0.12.0

mycroes, 2021-06-06:
So basically this needs retesting against 0.13.0 (for which I hope to get some feedback before releasing).

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes