S7NetPlus/s7netplus #299: Struct read/write

berkturkmen, 2020-08-24:
Hello,
I need a very urgent help. If you have please could you share me struct read and wtire example code?

FalcoGoodbody, 2020-08-26:
Hello @berkturkmen ,

I use to define a class in C# for any struct I want to read from the plc.
But I can give you an example how I did before:

 ```csharp
     public struct K2RakelLaminierstationState
        {
            public float SpaltAS;
            public float SpaltBS;
            public float FüllstandAS;
            public float FüllstandBS;
        }
         
     K2RakelLaminierstationState K2RakelLaminiereinheitStatus = new K2RakelLaminierstationState();
     K2RakelLaminiereinheitStatus = (K2RakelLaminierstationState)plcKonti2.ReadStruct(typeof(K2RakelLaminierstationState), 153, 2);

    ```

mycroes, 2020-09-04:
Closing, @FalcoGoodbody provided a concrete sample, I do hope to one day provide some proper samples as part of this repository.

closed 2020-09-04 by mycroes