S7NetPlus/s7netplus #305: Dotnet Core 3.1 - Linux Issue

bressallg, 2020-09-05:
I can't connect S7-1200 PLC when the program is running on Linux. No matter if the PLC exists or not in IP Address,
on Linux the error returns instantly.
Running the same project on Windows (console) works.
It seems that on Linux there is not a waiting time when you try to connect the PLC, while on Windows program it takes about 20s before return error, when there is no answer from the host.
Could you please help me to solve this issue?

scamille, 2020-09-06:
Interesting..I haven't really tested this on Linux yet but should be easy to do.
Unfortunately the integration tests are windows only using a snap7 server.

Anyway:
- are you sind the synchronous Plc.Open().function?

I suspect this might be related to some timeout setting. There should be nothing special in S7NetPlus besides standard .Net Tcp client socket calls, so no particular reason to not work on Linux.

scamille, 2020-09-06:
Try setting ReadTimeout = 0; and WriteTimeout = 0; before calling Open()

scamille, 2020-09-06:
Added linux UnitTest and fixed Timeout problem in #306

bressallg, 2020-09-06:
> Try setting ReadTimeout = 0; and WriteTimeout = 0; before calling Open()

Ok, working now! Thanks!

PR #318 (2020-09-16): Linux timeout fix & Multi-Platform Unit Test support
 Fix runtime error on linux with default Socket timeouts.

The default value for socket timeout API is 0 and not -1, so revert to that value to give the intended infinite timeout on all platforms.

 Test project: Enable netcore3.1 testing, switch win64 Snap7 and use custom port.

- Add netcoreapp3.1 target framework, allowing this to run on linux and macos as well.
- Switch windows snap7 DLL to 64bit version 1.4.2. This also improves UnitTest stability (reduces false positives) on the CI (including appveyor)
- Changing the port used for S7Ne [... cut]
closed 2020-09-16 by mycroes