S7NetPlus/s7netplus #310: Improve Feedback when reading unavailable DB or reading too many bytes

scamille, 2020-09-08:
When reading a DataBlock from the PLC that does not exist, or more bytes than exist in that DB, the error message is quite cryptic.

The idea is to figure out where/how in the S7 protocol such failures are communicated and turn that into a better Exception message.

scamille, 2020-09-08:
Reading a DB that does not exist gives me:
```"Invalid response from PLC: '32-03-00-00-00-00-00-02-00-04-00-00-04-01-0A-00-00-00'."```

scamille, 2020-09-08:
Reading past the end:
```Invalid response from PLC: '32-03-00-00-00-00-00-02-00-04-00-00-04-01-05-00-00-00'.```

scamille, 2020-09-08:
Current check in code is:
```
            if (s7Data[14] != 0xff)
                throw new PlcException(ErrorCode.ReadData,
                    $"Invalid response from PLC: '{BitConverter.ToString(s7Data)}'.");
```
So I think we can add a enum containing possible values for Byte 15 of the response data.
0xff would be OK
0x0A would be DB does not exist / not accessible
0x05 requests read too long.

And of course a bit more sophisticated logic to check that and give the appropriate error message.

mycroes, 2020-09-08:
https://github.com/mycroes/Sally7/blob/develop/Sally7/Protocol/S7/ReadWriteErrorCode.cs 😉

PR #316 (2020-09-16): Add more extensive response code validation.
Fixes #310
commit 10696416 (2020-09-16): Add more extensive response code validation.

Fixes #310
closed 2020-09-16 by mycroes