S7NetPlus/s7netplus #312: Connecting to a "S7 200 smart" PLC.

shuvendu2018, 2020-09-09:
Hello Expert,

Can you please share a example code(C#) to connect with "S7 200 smart" PLC. Does it require any configuration in PLC side.

FalcoGoodbody, 2020-09-09:
Hello @shuvendu2018 ,

I am not 100% sure but I think it is not possible to connect to a S7 200.

shuvendu2018, 2020-09-09:
Hi FalcoGoodbody,
Thanks for your quick reply.
I think "S7 200" and "S7 200 smart" are two different PLC model. I could not find any DB address in "STEP7 -Micro WIN/SMART" which is the programming software of "S7 200 smart". I want to read the "VW##" address from PLC.

I have gone through snap7 document and there I find that "S7 200" require tsap address instead of Rack and slot no. I am little confused about tsap.

FalcoGoodbody, 2020-09-09:
TSAP is nearly the same. If I remember correctly it looks like 03.02 for example.
03 defines that you are going to communicate with a pc or OPC-Server
02 defines the the rack 0, cpu in slot 2

For more information have a look here:
https://www.tanindustrie.de/pl/Help/ConfigClient/tsap_s7.htm

It is some years ago since I have got a S7-200 and never ever heard of a S7-200 Smart. But if you do not see any DB-Adresses of the variables within, it will be difficult to read the values.

mycroes, 2020-09-09:
Actually TSAP addresses are always used, but for most PLC's there's a 'free' connection that can be used from any source IP which is (apparently) based on slot and rack parameters. In TIA portal there's a network topology view where connections can be configured, you can also read the TSAP's there. I don't know how this works for STEP7-Micro. Also, I don't know what VW addresses are, but I'm not a PLC programmer. I'd be happy to help you in trying to get this going, but I'm a little short on information.

What you can try however, is to connect using any other PLC type but correct IP address and see what exception comes up, that might be a good starting point.

hiiWZY, 2021-02-26:
@mycroes  @FalcoGoodbody 
Hello, this is the S7-200 smart connection  request  info, I got it from WireShark and this worked very well for me.
`                
case CpuType.S7200SMART:
  bSend1[13] = 0x10;
  bSend1[14] = 0x00;
  break;
`

jaydeep6720, 2021-05-22:
My error while opening port to S7 200 SMART PLC

"PlcException: Couldn't establish the connection to 192.168.2.1.
Message: Unable to read data from the transport connection: An existing connection was forcibly closed by the remote host.."

My code for opening port

Plc _S7200 = new Plc(CpuType.S7200, "192.168.2.1", 0, 1);
      
        public MainWindow()
        {
            InitializeComponent();
            /*I open communication with the plc */
            _S7200.Open();

mycroes, 2021-06-02:
Hi @jaydeep6720 any chance you can test this with the code from #395?

jaydeep6720, 2021-06-03:
Dear Michel
my mobile number is +919375822001.
I have whatsapp enabled on this number.
you can send me links of code to test so i can test any Siemens code with
any Siemens PLC model.
THanks & regards
Jaydeep

jaydeep6720, 2021-06-03:
Test Results for changed Code:
Initialised code
Plc ROT = new Plc(CpuType.S7200, "192.168.2.1", 0, 0);

ABocde is not connected to S7200 SMart PLC having an onboard Ethernet port.
If i change PLC Cpu type to 300, it is very well performing all operations;

 Plc ROT = new Plc(CpuType.S7300, "192.168.2.1", 0, 0);

Also wants to inform you that all Siemens PLC uses integer as 16bit data
type, so update that part in library too to avoid data type confusion.
Thanks & regards
Jaydeep

BlackBoxRecorder, 2022-05-17:
For the V-region of the 200smartPLC, it is DB1.X, for example, V100=DB1.100, VW100=DB1.DBW100, try it.

To connect to 200smartPLC, rack=0,slot=0, try it.

asyura, 2024-04-24:
> For the V-region of the 200smartPLC, it is DB1.X, for example, V100=DB1.100, VW100=DB1.DBW100, try it.
> 
> To connect to 200smartPLC, rack=0,slot=0, try it.

so, how to read the other region ,like SM, M