S7NetPlus/s7netplus #313: Update docu writing datatyp StringEx question

FalcoGoodbody, 2020-09-11:
Hey there,

I was actually looking into the datatyp class of StringEx.
There is a method "ToByteArray" which converts a string into a S7-string with the neccessary 2-byte header.
Can someone show me if, where and how this can be used ? I could not find it in the code, sorry.
In the serialization class only String is used, but not StringEx.

I was wondering if with one can use this method, it would not be neccessary to do this by hand with a helper class like I descriped at docu.

Thanks for a short response.

scamille, 2020-09-11:
Ok so String looks more something for a Array of Char... / C-String variable, while StringEx corresponds to S7 String type with 2 byte header.

I think this should be first cleaned up in code before you bake it into the documentation. Some Serialization code does not support any string at all (Types.Struct for example)

I would go with the assumption of .NET string == S7 String. For "Char[] / C-String" we could still support it for doing manual serialization, or completely remove it.

FalcoGoodbody, 2020-09-11:
Yes....we could say that the normal string is only a Char[] which also exists in S7.
StringEx corresponds fully to S7-string.
Would be nice to have this (StringEx) functional in the library because this is an important use-case in my opinion.

scamille, 2020-09-11:
I personally have no strong preferences about Serialization stuff, so whatever makes more sense to correspond to a .NET string in S7 should be chosen. And I am also no SPS programmer, so no idea what a modern program should use. I have seen both used on SPS programs I transfer data from.

There is also the possibility of WString, which would be a more natural match for .NET string content since both are UTF-16 encoded, but I haven't seen that used at all, at least at my workplace.

FalcoGoodbody, 2020-09-11:
Well from the view of a plc programmer it is usuall use-case to get strings written in the plc and handle them correctly.
Often you get articel data into the plc like articel no., customer order-id or something and you have to track these data with the production process. 
If you get only a Char[] you have to do the conversion into a string niside the plc which is annoying. And you need correct strings if you need to display them a visualization system or something else. 
I myself prefer to get those data correctly from the system above.

So I would vote for supporting StringEx instead of Char[].

scamille, 2020-09-12:
I added #315 to rename StringEx to something more sensible.

string serialization seems available through Read/Write DataItems. As said it could be added struct/class serialization with the help of https://docs.microsoft.com/en-us/dotnet/api/system.componentmodel.dataannotations.stringlengthattribute?view=netcore-3.1 but I don't see this as high priority. Maybe someone else want to give it a try.