S7NetPlus/s7netplus #320: Error reading/writing data from PLC S71200

paulosscruz, 2020-09-14:
Hello, 

I am testing the communication of my software with a PLC of the model S7 12000, and I have problems reading or writing data.

This is my code:

```csharp
Plc plc = new Plc(CpuType.S71200, "10.160.47.195", 0, 1);
 plc.Open();
            
if (plc.IsConnected)
{
    var result = plc.Read(DataType.DataBlock, 100, 62, VarType.Int, 1);
    Console.WriteLine($"Read: {result}");
} 
```

This is the exception I get:

**StackTrace:**    em System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)
   em S7.Net.TPKT.Read(Stream stream) na C:\Users\paulo.cruz\source\repos\s7netplus-0.7.0\S7.Net\TPKT.cs:linha 36
   em S7.Net.COTP.TPDU.Read(Stream stream) na C:\Users\paulo.cruz\source\repos\s7netplus-0.7.0\S7.Net\COTP.cs:linha 55
   em S7.Net.COTP.TSDU.Read(Stream stream) na C:\Users\paulo.cruz\source\repos\s7netplus-0.7.0\S7.Net\COTP.cs:linha 105
   em S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count) na C:\Users\paulo.cruz\source\repos\s7netplus-0.7.0\S7.Net\PlcSynchronous.cs:linha 370
**InnerException:** System.Net.Sockets.SocketException (0x80004005): An existing connection was forced to be canceled by the remote host
   em System.Net.Sockets.Socket.Receive(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags)
   em System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)
**Source:** System

After debugging, I came to the conclusion that the exception is generated on the following line: https://github.com/S7NetPlus/s7netplus/blob/develop/S7.Net/StreamExtensions.cs#L27

I was also able to notice that there are tractive presence in the "ReadExact" method, is it possible that my problem is related to that?

Thank you!

scamille, 2020-09-14:
The exception comes from the underlying socket, saying that the connection was closed by the remote host (the PLC, or whatever you are connecting to here).
The recent changes should not have anything to do with it.

Did you follow the extra steps described in https://github.com/S7NetPlus/s7netplus/wiki/S7-1200-1500-Notes about GET/PUT and protection settings?

cc @FalcoGoodbody

paulosscruz, 2020-09-15:
First, thanks for the feedback!

Yes, all the extra steps informed in the documentation have been carried out.

Is there anything else I can do to isolate the problem and try to identify it?

An additional information for this, is that this exception is generated only when trying to read or write some data, after the connection with the PLC it appears as connected and available.

FalcoGoodbody, 2020-09-15:
Do you use "IsAvailable" method? This seems not to work acutally so please get rid of it and only use "IsConnected".
The data block is not optimized? 

Do you mind to show us the data block and the C# code?

scamille, 2020-09-15:
Stack overflow answer for your error message: https://stackoverflow.com/a/2582070

But I really have no good explanation why this is happening here. As @FalcoGoodbody mentioned check for the DB being unoptimized,.and maybe just try out a completely different DB as well.

mycroes, 2020-09-15:
@paulosscruz Are you getting the error on `Open` or on `Read`?

paulosscruz, 2020-09-15:
> Do you use "IsAvailable" method? This seems not to work acutally so please get rid of it and only use "IsConnected".
> The data block is not optimized?
> 
> Do you mind to show us the data block and the C# code?

@FalcoGoodbody , thank you very much!

The "IsAvailable" method does not seem to work correctly, after removing it the reading and writing functioned normally.

I appreciate the attention.

closed 2020-09-15 by paulosscruz