S7NetPlus/s7netplus #323: Optimized protocol

FrancescoCalugi, 2020-09-15:
Hi everyone,
more and more frequently I experienced Profinet PLCs with this optimized protocol setup, i.e. another type of addressing variables. Is there some code to read variables with this addressing type or at least some documentation about it?
Thanks

FalcoGoodbody, 2020-09-16:
Hi @FrancescoCalugi ,

with the S7-Net+ library it is not possible to read variable from an optimized data block.

mycroes, 2020-09-16:
Even more than that, we currently don't know what it would actually take to write optimized data blocks and if there's any open source implementation that actually supports this. It's something I want to investigate, but I've looked into it before and at that time couldn't figure a lot more than that it isn't supported in the protocol as used by S7NetPlus but actually requires a whole lot more complicated protocol, for which there seems to be no public documentation available.

FrancescoCalugi, 2020-09-16:
I supposed that, please let me know if you find something new about it. Thanks a lot.

AlessandroMinunnoNOV, 2022-01-20:
Hi, how is going with this issue? There are any news to read and write when the blocks are optimized?
Thanks

Jason-Jelks, 2022-01-20:
According to everything I have worked with, read in Siemens documentation and discussed with Siemens Engineers, it is not possible to read or write to optimized data blocks externally. I have not heard of any plans for Siemens to change that either.

gfoidl, 2022-01-20:
There's a non-Siemens propietary solution that can access optimized data blocks. They did this by reverse engineering the protocol (without violating any IPs they said). One can use your favorite search engine to find that -- I don't want to make ads for them here by posting the link.
But it's not open source anymore (and we all embrace open source here...)

If viable one could use OPC UA instead of the S7-protocol, so that symbolic names can be used.
Although that comes with other concers along the way.

mycroes, 2023-11-21:
I have asked Siemens (Netherlands) if they're willing to share any information on the optimized protocol. Siemens is proclaiming to be more open nowadays (I intended an event titled _Open Industrial Automation_ at Siemens recently), but I don't have any expectations.

U11Leung, 2024-01-04:
As far as I know, kepserver is one among the softwares that can do symbolic addressing on 1200/1500, as well as this repo: https://github.com/thomas-v2/S7CommPlusDriver . Although kepserver is not an open source solution, still it proves it is not impossible to do that.
Another thing to notice is the above repo has decent revision FW requirement on CPUs, but kepserver can talk to much older CPUs. Kepserver describes its data driver as "Siemens S7 Plus Ethernet" driver.

asyura, 2025-09-19:
This one also can do this
https://www.indi-an.com/en/plccom/for-s7/fors7-overview/