S7NetPlus/s7netplus #324: Feature Request/Improvement/Enhancement: WriteMultiple

FalcoGoodbody, 2020-09-15:
When testing some functions to update the documentation I became aware of that there is really good working "ReadMultiple"-method but no "WriteMultiple".
One can use "Write"-method to write multiple variables to the plc. But at "Write"-method you need to provide an array of DataItem. At the "ReadMultiple" you need to provide a List<DataItem>.

As I am a not a soo advanced C# progammer it would be more intuitive to me if there would be a "WriteMultiple"-method or if the "Write"-method would accept a List<DataItem> so that it would be more similar to the "ReadMultiple"-method.

mycroes, 2020-09-16:
Hi @FalcoGoodbody, basically it comes down to this; when you do a write on the protocol level there's no distinction between writing a single value or a multitude of values. As such, the `Write` method accepts a `params DataItem[] items`, which means you can pass in as many items as you want as arguments (that's what `params` is for). Honestly there's no good reason why `ReadMultipleItems` doesn't use similar practice, but I guess that comes down to the fact that the original authors are _not so advanced C# progammers_ either 😉

I'm actually the author of the `Write` method that accepts multiple DataItems, because it wasn't there when I started using S7NetPlus. As you've seen I've been the only maintainer lately (but happily we have some good contributors lately!), but back when I wrote that code there was a strong desire to keep backwards compatibility in the library. With recent improvements by others (like @scamille) I noticed that the backwards compatibility is becoming more of an issue than a solution, so I've since taken the stance that we should improve the codebase regardless of the backwards compatibility, as long as usability is just as good as it always has been. Another issue there is the lack of documentation, but hopefully we can improve that as well.

Anyway, long story short: I think the `Write` method is fine as it is, although we should add an overload with `IEnumerable<DataItems>` for usability (because often you already have some type of enumerable that you want to read). I also think we should actually have similar `Read` methods to have a coherent API. Would this satisfy your request?

scamille, 2020-09-16:
More abstractlly said we should have a symmetrical API between read and writes and also between async and synchronous functions.

I am sure we will get there eventually, I can take a look at it after current changes are integrated.

mycroes, 2020-09-16:
That'd be great. Obviously I also don't mind doing it, but time is usually the issue. I intend to catch up on PR's again this evening.