S7NetPlus/s7netplus #325: wrong addressing when using Write() to write multiple data items

FalcoGoodbody, 2020-09-16:
given is a data block with the following variables
[image]

in C# I added for each variable a dataItem
```csharp
private static DataItem varWordWrite = new DataItem()
  {
      DataType = DataType.DataBlock,
      VarType = VarType.Word,
      DB = 83,
      BitAdr = 0,
      Count = 1,
      StartByteAdr = 146,
      Value = new object()
  };

private static DataItem varIntWrite = new DataItem()
  {
      DataType = DataType.DataBlock,
      VarType = VarType.Int,
      DB = 83,
      BitAdr = 0,
      Count = 1,
      StartByteAdr = 148,
      Value = new object()
  };

private static DataItem varDWordWrite = new DataItem()
  {
      DataType = DataType.DataBlock,
      VarType = VarType.DWord,
      DB = 83,
      BitAdr = 0,
      Count = 1,
      StartByteAdr = 150,
      Value = new object()
  };

  private static DataItem varDIntWrite = new DataItem()
  {
      DataType = DataType.DataBlock,
      VarType = VarType.DInt,
      DB = 83,
      BitAdr = 0,
      Count = 1,
      StartByteAdr = 154,
      Value = new object()
  };
```

I assigned values to the data items
```csharp
// asign values to the variable to be written                
    varWordWrite.Value = 67;
    varIntWrite.Value = 33;
    varDWordWrite.Value = 444;
    varDIntWrite.Value = 6666;
```

I use `Write()` to write the items to the plc
```csharp
DataItem[] write = new DataItem[3];
write[0] = varWordWrite;
write[1] = varIntWrite;
write[2] = varDIntWrite;

myPLC.Write(write);
```

As you can see in the screenshot the values are not written correctly to the plc. The first 2 items are not written to the their startaddress, instead they got shifted 2 words and are written into the DWORD at startaddress 150.
I already tried to only write the word variable that way and I can say that it is written at the startadress if the INT at 148.

scamille, 2020-09-16:
As far as I understand, S7NetPlus uses the Type of the value object to serialize DataItems, and mostly ignored the VarType. See https://github.com/S7NetPlus/s7netplus/blob/develop/S7.Net/Protocol/Serialization.cs#L14

So because you are writing ints to all those variables instead of ushort (=Word), short (=Int), unsigned (=DWord) and int (=DInt), you get these results.

I can't really say if this behavior is desirable, but that's what it is.

FalcoGoodbody, 2020-09-17:
Well I also did a mistake here when asigning the values to the data items.
Instead of:
```csharp
varWordWrite.Value = 67;
varIntWrite.Value = 33;
varDWordWrite.Value = 444;
```
one should do the correct conversion:
```csharp
varWordWrite.Value = (ushort)67;
varIntWrite.Value = (ushort)33;
varDWordWrite.Value = (uint)444;
```
and then everything is fine.

Thank you for your advice

closed 2020-09-17 by FalcoGoodbody