S7NetPlus/s7netplus #326: wrong addressing when using Write() with single variable

FalcoGoodbody, 2020-09-16:
I just testet to write a single value using
```csharp
myPLC.Write(DataType.DataBlock, 83, 146, 100);
```
and also tried
```csharp
myPLC.Write("DB83.DBW146", 111);
```
The value is always written into DB83.DBW148 instead of DB83.DBW146.

scamille, 2020-09-16:
I can't really understand your strange string-based addresses as a non-PLC programmer. (They are also not really documented, right?)

Can you write an example using the first version of Write that you have (```myPLC.Write(DataType.DataBlock, 83, 146, 100);```) which matches the displayed DataBlock?

FalcoGoodbody, 2020-09-17:
My fault guys...sorry.
I was a little bit confused yesterday.
Everything works fine until you use the correct value conversion.
If you write `myPLC.Write("DB85.DBW4",22)` the 22 is interpreted as an C# INT which is a DInt in S7.
If you write `myPLC.Write("DB85.DBW4",varValue)` and varValue is a short everything is correct.

It is the same when using ´myPLC.Write(DataType.DataBlock, 83, 146, 100);´ The 100 is a C#-Int and therefore a S7-DInt.
Strongly recommendet to use variables instead of numbers. All this is already described in the docu but I was not able to read correctly. Sorry.

closed 2020-09-17 by FalcoGoodbody

scamille, 2020-09-17:
Maybe S7.NetPlus should validate that the var type with the type of the value before doing anything.

FalcoGoodbody, 2020-09-17:
Maybe...would be a nice feature