S7NetPlus/s7netplus #327: Investigate MaxPDUSize calculations.

scamille, 2020-09-17:
As discussed in #314 there might still be something wrong with how the maximum PDU size is calculated when establishing a connection.

This ticket is intended to split this off into a separate ticket.

Main relevant part so far from the other issue:
> The byte identifies PDU sizes (instead of giving a PDU value), this is from the RFC:
> 
> ```
> 0000 1101  8192 octets (not allowed in Class 0)
> 0000 1100  4096 octets (not allowed in Class 0)
> 0000 1011  2048 octets
> 0000 1010  1024 octets
> 0000 1001   512 octets
> 0000 1000   256 octets
> 0000 0111   128 octets
> ```
> 
> Now take the decimal values of above, and perform 2 ^ [decimal value]:
> 
> ```
> binary     size          dec.   2 ^ dec.
> 0000 1101  8192 octets   13     8192
> 0000 1100  4096 octets   12     4096
> 0000 1011  2048 octets   11     2048
> 0000 1010  1024 octets   10     1024
> 0000 1001   512 octets    9      512
> 0000 1000   256 octets    8      256
> 0000 0111   128 octets    7      128
> ```
> 
> I actually looked into Sally7 to verify this and I must admit I was using the same logic as applied here (UInt 16 from 2 bytes), so honestly I have no clue whether the S7 protocol adheres to the RFC. I guess this is something to investigate.
> 
> The Connection Request is actually different from Data Transfer (which I believe is used for read/write), Connection Request mostly or entirely adheres to the RFC while read/write uses a custom protocol inside DT messages.
_Originally posted by @mycroes in https://github.com/S7NetPlus/s7netplus/pull/314#discussion_r489749807_

FalcoGoodbody, 2020-09-28:
@scamille : may I help here with testing or logging the data when establishing a connection (WireShark or so? )

scamille, 2020-09-28:
Yes you might be able to test various PLC types. But this would require a testable version with the changed PDU size calculation.

The test would then be to just see if a large read/write calculation works or not.

I'll see what I can do.

FalcoGoodbody, 2020-09-28:
I can test S7-1516F, S7-300 and S7-1500SP plc. If I can help and test, please don´t hesitate to contact me. Would be nice to have a little explanation then for the test

scamille, 2020-09-28:
Well the MaxPdu size is simply a limit used for creating S7 messages. It is negotiated during connection establishment, and then used for the remaining session when eg. reading data.

If the calculated MaxPDU Size is too large, the PLC will probably just respond with an error.