S7NetPlus/s7netplus #328: My logging application stops reading values in 0.7.0

SmackyPappelroy, 2020-09-17:
I updated my logging application from 0.1.9 to 0.7.0. 
After running the application for 2 days I received Exceptions 
[image]

I had to rewrite my code whenever I receive a PlcException with ErrorCode == ErrorCode.WrongNumberReceivedBytes I had to close and open a new connection, to be able to program around this problem.

Does anyone have any ideas why this is happening.

scamille, 2020-09-17:
I assume the first one is happening because of some networking problem, so S7.NetPlus runs into a timeout. That is to be somewhat expected when doing network communication.

What should not happen of course is that after the  initial exception you can not continue doing normal operations. The connection should be either fully functional, or if we can not recover S7.NetPlus should tear down the whole thing down so you can do a graceful reconnect.

Or in other words a simple:
'''if (!PLC.IsConnected) PLC.Open()``` before any periodic read/write calls should be all that is needed from user code.

SmackyPappelroy, 2020-09-17:
I fully agree with what you wrote! Do you have any idea what the first exception might be? 
If so I can catch it an do a reconnect.

scamille, 2020-09-17:
IsConnected definitely does no longer open a new socket. (IsAvailable still might, but you should not use that).

@mycroes I suspect that we need to do something extra in cases where we receive invalid data which does not match the underlying tpkt / cotp protocols, or just generally when receiving EOF/timeouts on the stream.
Either flush the stream or tear i down completely. The first one is definitely less invasive, but I don't know how reliably it resets a stream which has its state messed up.

@SmackyPappelroy as said, I would not worry too much about your initial exception. The one on the screenshot can have multiple causes, either because of something in the network, PLC not replying correctly or an error in S7.NetPlus.

SmackyPappelroy, 2020-09-17:
First of all thank you very much for taking time to help my problem. 
My issue is that I want to have an exception when I can't read or write. If I check for "IsConnected" at all read or writes, and try reopening the connection the actual error might be lost in the application.
I want to have an exception, so I can act accordingly. 
My suggestion is that the developers might expand the PlcException class with this exception and maybe add another ErrorCode for this, so it is easier to interpret.

FalcoGoodbody, 2020-09-18:
@SmackyPappelroy can you please show your code how do you access the PLC exception and error code? I would like to implement it as well. Thanks in advance

SmackyPappelroy, 2020-09-18:
@FalcoGoodbody  I just looked at my logging application and my fix has not worked and was not able to catch the correct exception.
I had to revert back to S7.NET version 0.1.9, and try to replicate the problem in 0.7.0.
I will come back when I find a solution that works.

FalcoGoodbody, 2020-09-21:
@SmackyPappelroy I just find out how to address the PlcException.ErrorCode.

```csharp
// open the connection to the plc
try
{
    myPLC.Open();
}
catch (PlcException e)
{                    
    Console.WriteLine(e.ErrorCode.ToString());
}
```

closed 2020-09-30 by SmackyPappelroy