S7NetPlus/s7netplus #329: Cant Read S71500

RubenFarfan08, 2020-09-18:
hi,
i am testing with s71500

but when i read variables i cant
[image]

[image]

[image]

and i use the guide for S71500
https://github.com/S7NetPlus/s7netplus/wiki/S7-1200-1500-Notes

but i cant read db.

please let me know if i'm coding wrong
Thanks

FalcoGoodbody, 2020-09-18:
@RubenFarfan08 please check the also the conversion page in the documentation and read carefully the S7-1200-1500 notes. As I can see on your screenshot you are trying to read from an optimzed data block which is not supported by this library.
Please change the attributes to non-optimized, activate PUT/GET communication and set the access level to full. Follow then the docu and the conversion page.

If you might still have problems, don´t hesitate to ask for help here.

FalcoGoodbody, 2020-09-24:
@RubenFarfan08 could you solve your problem? Does it work now?

SmackyPappelroy, 2020-09-24:
@RubenFarfan08 Youre problem might be that your DB is optimized. Uncheck that tick-box in TIA portal and try again

wangzhongkui, 2021-04-18:
Read(DataType, 78, 0, VarType.Dint,16,)

closed 2021-06-02 by mycroes