S7NetPlus/s7netplus #330: How to access the slack-channel?

FalcoGoodbody, 2020-09-18:
Is the slack still active? Do I a need an invitation to get access? 

Could someone please bring me update? ;-) 
I am at the MakerVsVirus workspace already and would like to enter the S7Net+ workspace if it exists. 
Maybe we could communicate there a bit more faster?

Thanks in advance

mycroes, 2020-09-21:
No, it's not active (it still exists though), I'm not sure if you need an invitation, but it should be at http://s7netplus.slack.com. However, as mentioned, it's not active. There doesn't seem to have been anyone there for over a year. I'm also not a Slack user, so I only ever used it for S7NetPlus back in the days when we setup the GitHub organization. Not sure if it's faster to communicate, it might be a better fit for discussions like this issue.

scamille, 2020-09-21:
We could also use the discussion functions in github, assuming we would get membership to the organization (which does not mean you have to grant any team membership with permission for now).

I also do not use slack, and don't think I would add yet another communication app just for this project.

We can also just keep discussing in public issues. There is no reason not use them for strategic discussions as well.

FalcoGoodbody, 2020-09-22:
Ok thank you for your answers.
It seems as if it is better to keep everything at is and to use discussions.

closed 2020-09-22 by FalcoGoodbody