S7NetPlus/s7netplus #331: Plc.IsAvailable{get;}  cause disconnection with plc.

SheepChao, 2020-09-24:
bool PlcOK=Plc.IsAvailable;

this could cause disconnection with the plc.

FalcoGoodbody, 2020-09-24:
This is a known behaviour. The maintainer of the library are working on this.
I think in future version it will not be available annymore.

Please do not use `Plc.IsAvailabe` and use `IsConnected` instead

ulichx, 2020-11-02:
> 
> 
> This is a known behaviour #186 . The maintainer of the library are working on this.
> I think in future version it will not be available annymore.
> 
> Please do not use `Plc.IsAvailabe` and use `IsConnected` instead

`IsConnected `doesn't work the same as `IsAvailable`. If the connection drops for any reason (eg: I remove the ethernet cable for testing) **after I connect to a plc**, there is no way to check it without `IsAvailable`. Without checking the current status of connection, `IsConnected `always returns `true `until I `Close()` the connection. 

For me the workaround solution is; inside `if(Plc.IsAvailable){}`  and before I read/write something to PLC, I do `Plc.Open()` again.
It would be great though, if IsConnected was aware if the connection has dropped or still solid.

SmackyPappelroy, 2020-11-02:
if you get an exception from the plc object that is related to the connection, you could just try and ping the plc and then connect. That way you know if the ethernet cable is removed @ulichx

ulichx, 2020-11-02:
@SmackyPappelroy I will try it as soon as possible. No matter what `IsAvailable` gives me headaches. I avoid it as much as I can. Sometimes my workaround works, sometimes `Open()` after `IsAvailable` gives an error. After some repeated `try{}`s sometimes it connects, sometimes doesn't etc.

So, I can ping the PLC while it's connected, to check if it's still a solid connection, right? It doesn't break the connection like `IsAvailable`?

SmackyPappelroy, 2020-11-02:
IsAvailable opens a new NetworkStream and therefore breaks the connection! I don't know why it is still left in the code! A ping is harmless! It doesn't say anything about the connection, but at least you know that the cable is connected

closed 2021-06-02 by mycroes