S7NetPlus/s7netplus #332: Connect S7-200 TPKT header is incomplete / invalid. Received Bytes: 0 expected: 4

hongphucbk, 2020-09-24:
I try connect to S7-200 using ethernet module CP 243-1 but has error.

[image]

[image]

I think default TSAP in S7-200 is difference with S7.net 
[image]

Please give me advise? Thank you

FalcoGoodbody, 2020-09-24:
please see #322 
This problem should be solved in Version 0.7.0.
Can please update to the newest one and retest?

mycroes, 2021-06-02:
Late to the show, but yes, I think your assesment is correct. Any chance you can test the code from #395?

ooowuyaooo, 2021-06-03:
> Late to the show, but yes, I think your assesment is correct. Any chance you can test the code from #395?

Hi bro 
you means here the lcoal and remote TSAP should be the same as 10.01 ? if set other values it can not be connected?

mycroes, 2021-06-03:
Hi @ooowuyaooo,

The screenshot from @hongphucbk shows the TSAP's are 10.01 and 10.00, if I understand the image correctly the PLC side is 10.01 and local 10.00. These values are quite essential in the S7 protocol, if they're incorrect the PLC will just close the connection (as evidenced by the exception). If the TSAP's from the screenshot are correct, the remote TSAP should be 10.01 (or `0x10`, `0x01`) instead of the previous value. I don't have such a PLC myself, so I can't test it myself either...

Regards,
Michael

ooowuyaooo, 2021-06-03:
Hi @mycroes 
First thank you for you codes it really helps me a lot.
yes i mean that TSAP values can be setted at will such as (local 12.01 remote 12.02), so are there availble to add this TSAP to the siemens s7-200 connection string ? like   Plc s7200plc = new Plc(CpuType.S7200, "10.136.176.3", 0, 2,"local tsap","remote tsap");
now everytime with the different TSAP i have to make lots of the dll file...
thanks bro 
regards
edis

mycroes, 2021-06-03:
@ooowuyaooo Yes, I'll add an option to specify TSAP's.

ooowuyaooo, 2021-06-03:
@mycroes 
OK thanks
one thing i am not sure you know or not is  what we talked in this topic are all with siemens s7-200 & CP243 module (because old type s7-200 doesn't has the ethernet port so it need CP243 to extend one ethernet port ) not the s7-200 Smart(it has the ethernet port on the cpu)..
the s7-200 Smart works well with the current string. 
regards,
edis

mycroes, 2021-06-04:
Hi @ooowuyaooo,

> yes i mean that TSAP values can be setted at will such as (local 12.01 remote 12.02), so are there availble to add this TSAP to the siemens s7-200 connection string ? like Plc s7200plc = new Plc(CpuType.S7200, "10.136.176.3", 0, 2,"local tsap","remote tsap");

#396 now adds support for the following: `var plc = new PLC("192.168.1.10", new TsapPair(new Tsap(0x10, 0x00), new Tsap(0x10, 0x01));`. Could you give that code a spin to see if it works for you for S7 200? Rack and slot are normally only used to calculate the TSAP addresses, which in all cases except `Logo0BA8` and `S7200` are actually equal. Per [@jaydeep6720 in PR 395](https://github.com/S7NetPlus/s7netplus/pull/395#issuecomment-853634190) I've actually decided to add S7 200 Smart which will use the same calculation as S7 300, 400, 1200 and 1500. I'm actually starting to wonder if it's not just that calculation or manual set addresses, but I have no idea about `Logo0BA8`.

I do take it that S7 200 doesn't have a default? You always need to choose TSAP yourself? If so I can remove it from the calculation method so people will know to use the alternate constructor.

@hongphucbk Maybe you could also give the code in #396 a try?

Regards,
Michael

jaydeep6720, 2021-06-05:
Dear Michele
I want to add one information as you knew already.
Connecting to S7200 (With CP 243-1) & S7200Smart may differ as i don't have tested with S7200( CP243-1).
May be we can give a trial to connect.
CP243-1 is a bit costly part so in solution Siemens has launched S7200Smart which no needs of CP for Ethernet access. 
S7200 is Conventional PPI plc & CP243-1 needed to be initialized with proper parameters.

@ooowuyaooo Why are you making Slot as 2 ? Check if you can connect with S7300 Type with Slot 0 Rack as 0.
Siemens S7200 System conventionally never configured for different Slots & Rack. Actually Slot & Rack parameters will be neglected by PLC for any 3rd Party communication (its based on my Experience).
Final trial  code for Connecting S7200 :

 Plc s7200plc = new Plc(CpuType.S7300, "10.136.176.3", 0, 0);

Let TSap overloadd with default by class.
check if you can succeeded !!

mycroes, 2021-06-05:
> @ooowuyaooo Why are you making Slot as 2 ? Check if you can connect with S7300 Type with Slot 0 Rack as 0.
> Siemens S7200 System conventionally never configured for different Slots & Rack. Actually [...]

I'm not a 100% sure, but from what you can find online the rack and slot numbers definitely matter. I also read that for S7 300 the slot number is always 2.

You're partially correct about the fact that the PLC's don't care about rack or slot for S7 communication, they do however care about the TSAP addresses during the Connection Request phase, which is exactly the only thing that S7NetPlus uses rack and slot for.

jaydeep6720, 2021-06-05:
Dear Bro
I said S7200 doesn’t bother about rack slot.
Actually Slot & rack no’s are for S7 300 & above PLCs as these are PAC plcs means they required Backplane bus for communication compulsory.
Due to this backplane architecture , they support High IO (input output) capacities especially Analog ios.
S7200 is modular series PLC means all Input output addresses are fixed list & supporting IO capacities are also less infect only 256 IO capacities with 32 Analog IO max.

ooowuyaooo, 2021-06-07:
@mycroes 
in S7-200 Micro-win Ethernet setup for the CP243 is like this , in "Peer-to-Peer connections" page you can choose how many connections you want (0~8), then goes to next page to set the Remote tsap manually , the Local tsap is grey(sometimes 10.02 or 11.02 or 12.02...) i don't know why it cannot be modified ,and if you choose you need 8 connections you will get 8 different Local tsap (defaulted ,such as 10.02/11.02/12.02/13.02/14.02...) then you should configure 8 different tsap . 
the Rack and SLOT for the s7-200 with CP243 is no matter what you configured. 
sorry for the LOGO0BA8 i have no idea what is that.

ooowuyaooo, 2021-06-07:
@mycroes 
[image]
[image]
in S7-200 Micro-win Ethernet setup for the CP243 is like this , in "Peer-to-Peer connections" page you can choose how many connections you want (0~8), then goes to next page to set the Remote tsap manually , the Local tsap is grey(sometimes 10.02 or 11.02 or 12.02...) i don't know why it cannot be modified ,and if you choose you need 8 connections you will get 8 different Local tsap (defaulted ,such as 10.02/11.02/12.02/13.02/14.02...) then you should configure 8 different tsap .
the Rack and SLOT for the s7-200 with CP243 is no matter what you configured.
sorry for the LOGO0BA8 i have no idea what is that.

mycroes, 2021-06-07:
Hi @ooowuyaooo,

Don't worry about the other types. Are you able to test the code from #396 with S7-200? It's already merged to develop as well, so if you have a local clone of the git repo you could just pull and run. For your example in the second screenshot you can use the following code:
```csharp
var plc = new PLC("10.136.176.3", new TsapPair(new Tsap(0x10, 0x01), new Tsap(0x10, 0x02));
```

Regards,
Michael