S7NetPlus/s7netplus #334: My logging application reads wrong values from PLC after a while

SmackyPappelroy, 2020-09-28:
I have a logging application that uses S7Net 0.7.0.
I log a number of values from different PLCs. After running my application for at least an 30 minutes, the values I recieve using: 

public object? Read(DataType dataType, int db, int startByteAdr, VarType varType, int varCount, byte bitAdr = 0)

are not the correct tags. For instance if I log one tag in "DB0.DBW0" and another in "DB0.DBW2" the values are switched. So if DB0.DBW0 is 0 and DB0.DBW2 is 2 I receive values for 0 when reading DB0.DBW2 and 2 when reading DB0.DBW0. 

This only happens after running for at least 30 mins. As a note the network is not 100% stable in which the PLC's are connected.
I have tried to replicate this in a simulated environment without success. Does anyone know if is a bug in S7.Net or what it could be.

I am attaching a trend-curve for my logging. The curve should be straight, as the value is always increasing, but it is not.
[image]

Here is a snippet of my code, for reading the PLC values:
[image]

FalcoGoodbody, 2020-09-28:
@SmackyPappelroy can you show us the code from the plc where the values are written to the DB0? What type of plc do you use?
Did you check the values in the plc after these 30 minutes? Can it be possible that an DBD is addressed inside the plc for these values?

SmackyPappelroy, 2020-09-28:
Hello @FalcoGoodbody!
The PLC-code is correct. For the provided trend curve, that value is always increasing. 
The values I am seeing when it is wrong, is other tags I am logging.
Only at 12:00 am the value is set to zero for the trend curve tag. The problem is not lying in the PLC. It is either wrong in my code or something wrong with the S7.Net read function.

FalcoGoodbody, 2020-09-28:
hmm...In your code you are trying to read a String?
What is this function `MyPlc.ReadS7StringtoString`? I do not know that such a construct exists.

Did you try to read the values with for example `ushort result = (ushort)plc.Read("DB1.DBW0");`

SmackyPappelroy, 2020-09-28:
I have a method that pings the Plc at 10 second intervals. If I can't ping I do a Plc.Close() and then a Plc.Open().
If Plc.IsConnected() == true I resume my logging.
But I believe that the problem might be that even though I can't ping the Plc, Plc.IsConnected() == true.
Thats why my data is corrupted. If this is the problem that is a major issue. Can someone please test this!

FalcoGoodbody, 2020-09-28:
I put all my `myPlc.Open()`code in a try-catch and handle the `PlcException`.

I would like to test it for you but unfortunately I can not disconnect the plc´s in production.
How should someone test the behaviour you described?

SmackyPappelroy, 2020-09-28:
I have a S71500 plc available, and will do a test this evening. I will get back to you with the results.

FalcoGoodbody, 2020-09-28:
I just looked into the code and found at `PLC.cs´ line 115 that there is some work to do.
The comment say "TODO: actually check communication by sending an empty TPDU
So at the moment there is no real check.
I googled a bit and found maybe a first step which could be integradet:
```csharp
//TODO: Actually check communication by sending an empty TPDU
public bool IsConnected
{
    get
    {                
        try
        {
            bool clientConnected = false;

            if (tcpClient == null)
                return false;

            //TODO: Actually check communication by sending an empty TPDU
            if (tcpClient.Client.Poll(0,SelectMode.SelectRead))
            {
                byte[] buffer = new byte[1];                        
                if (tcpClient.Client.Receive(buffer, SocketFlags.Peek) == 0)
                {
                    // client disconnected
                    clientConnected = false;                            
                }
                else
                {
                    clientConnected = true;                            
                }
            }

            if (clientConnected)
            {
                return tcpClient.Connected;
            }
            else
            {
                return false;
            }

        }
        catch { return false; }
    }
}
```

If this ok we could do a Pull-Request

SmackyPappelroy, 2020-09-28:
@FalcoGoodbody Great!
I will make a Unit test with this code this evening. 
I will also make a test with the old code. I will get back to you with the results of the tests

scamille, 2020-09-28:
I don't think this has anything to do with the underlying problem.

IsConnected can always be only a hint if the connection is alive or not. Even with your improved version, when you start the Read after calling IsConnected with a successful return value, the connection might already be terminated by your operating system or the PLC.
So you always have to assume that your read/write can fail because the connection was no longer available, and if you require a successful execution you should retry the whole thing.

Without seeing the actual code it is really hard to know what goes wrong here

SmackyPappelroy, 2020-09-28:
I was able to see the exception thrown from the Plc class when the data started to be corrupted. As I mentioned before the network on which the Plc's is, is very bad, so thats probably why not anyone has found this error before. 
The exception I receive is and PlcException with ErrorCode.WrongNumberReceivedBytes
What I do now in my application is that I close the Plc and dispose of the Plc object and create a new Plc.
This works, but is a very bad solution. I am thinking of going back to version 0.1.9 because I never had these problems when using that version. Does anyone have an idea what to do when the network stream data is corrupted?

FalcoGoodbody, 2020-09-29:
Well I check all my readings and writings from and to the plc and and if an PlcException occurs, I only close the connection and open it again before the next reading or writing.
There where a lot of improvements in the last versions. I started with using 0.4.0 last year and I am now on 0.7.0.
I have to admitt that I am very satisfied with version 0.7.0.
Yes there are now a lot more situations when you get an exception which was not before but, in my opion, that helps me more than not having it.
I did spent more time in correct error handling now and learned a lot because of this. My application is now more stable than ever before.

If you know that the bad network is the reason for your problems, why you don´t start at this point and try to make the network better instead of trying to handle this in your application?

SmackyPappelroy, 2020-09-29:
Thank you @FalcoGoodbody for good feedback. 
I will try your approach and closing and reopening the connection as soon as I receive a PlcException.
I will run my application today and see if that fixes the problem.
Regarding the network, they customer is changing the switches and troubleshooting the network, but it is a loooong process, and the logging has to work in the meantime.

SmackyPappelroy, 2020-09-29:
@FalcoGoodbody My application has been running steadily for  10 hours now! When I get a PlcException I close the plc and wait a few seconds  (so the network has time to stabilise) and then open the plc again! This seems to be working so far 🤞

FalcoGoodbody, 2020-09-30:
@SmackyPappelroy congatulation! I wish you the best and hope it is still running.
I made my last changes in error- and exception handling 28.09.2020 3:30 pm and the program is still running. Logs every 10 seconds more than 300 values/structs/classes. Actually the 17.675 entry was done since the restart.
Looks very stable at all

@mycroes I think this issue can be closed.