S7NetPlus/s7netplus #336: Close connection on Stream EOF or other exception

scamille, 2020-09-30:
@mycroes There have been some issues recently (#334 and #328) where the user had trouble to continue operating after communication problems occurs.

The solution from @FalcoGoodbody was to manually force Close() after certain exception, which seems to work fine for him.

The reason I am opening this issue is that I don't think we should burden the user with having to manually closing the PLC connection on exception cases. All the user should be required to do is have a ```if(!plc.IsConnected) plc.Connect();``` line before doing stuff, in my opinion.

My current feeling is that we should definitely close the exception on any stream EOF. In that case we can probably not really recover from the faulty state with the next request.

Maybe we should even close the connection on any IOException? But I am not really sure if that is overzealous and has too many false positives, where the connection would still have been ok to work with.

FalcoGoodbody, 2020-10-01:
Hello together,

I just want show a whole method I use to read values from one plc.
There is everything included which made the application real stable.

```csharp
private void ReadPlcKonti1()
{
    string EintragLog;
    try
    {
        if (!plcKonti1.IsConnected)
        {
            plcKonti1.Close();
            plcKonti1.Open();
        }

        if (plcKonti1.IsConnected)
        {
            K1Zaehlwert1 = ((uint)plcKonti1.Read("DB202.DBD0")).ConvertToInt();
            K1Zaehlwert2 = ((uint)plcKonti1.Read("DB202.DBD4")).ConvertToInt();
            K1Geschwindigkeit = ((uint)plcKonti1.Read("DB202.DBD8")).ConvertToFloat();
            K1aktuelleLaufmeter = ((uint)plcKonti1.Read("DB202.DBD14")).ConvertToFloat();

            #region Rollen
            var statusProduzierteRollen = plcKonti1.ReadBytes(DataType.DataBlock, 203, 0, 1);
            NeueRolleKonti1 = statusProduzierteRollen[0].SelectBit(0);

            if (NeueRolleKonti1)
            {
                plcKonti1.ReadClass(prodRolleKonti1, 203, 2);
                var tstring = plcKonti1.Read(DataType.DataBlock, 204, 0, VarType.StringEx, 50, 0);
                AuftragsnummerKonti1 = tstring.ToString();
                plcKonti1.WriteBit(DataType.DataBlock, 203, 0, 0, false);
            }
            #endregion
        }                
    }
    catch (PlcException e)
    {                
        // wenn dies nicht hilft, dann Log-Eintrag und weiter
        if (!CheckVerbindungPlcKonti1())
        {
            plcKonti1.Close();
            EintragLog = " Verbindung zur SPS Konti1 wurde unterbrochen...Zeile 2003";
            UpdateListView(lstBxLog, AnzahlDatensatz, EintragLog, false);
            dataLogger.Info(EintragLog);
        }

        if (e.ErrorCode == ErrorCode.WrongNumberReceivedBytes)
        {
            plcKonti1.Close();
        }

        if (e.ErrorCode == ErrorCode.ReadData)
        {
            plcKonti1.Close();
        }

        EintragLog = e.ErrorCode.ToString();
        UpdateListView(lstBxLog, AnzahlDatensatz, EintragLog, false);
        dataLogger.Info(" Konti1 - ErrorCode: " + EintragLog);
    }
}
```

SmackyPappelroy, 2020-10-01:
Do you think there is any negative effects, if s7.net always plc.close() after a PlcException? @scamille

scamille, 2020-10-01:
For your use case, it should be fine. All it means is that you are closing the connection and connecting again, even if you don't have to. All that does is add some delay for the reconnecting. If your exceptions because of network problem are rare, that shouldn't really make any difference since during normal operation this will never occur.

This ticket is intended for a solution in the S7.NetPlus library itself. Here the goal should be to get much closer to the optimal - only close the connection if it is corrupted and we (the library, the connection or the PLC) can not recover without closing and doing a clean reconnect.

SmackyPappelroy, 2020-10-01:
That was what I meant! Why the library does not close the connection if you have an exception from the networkstream!

rjanak, 2020-10-13:
Not sure if my issue is related to this, I'm doing multiple calls to the PlC, it works fine most of the time but it stops working for some calls but other calls are fine. I've added a client.Close() Client.Open clause when the client.IsConnected is false. The exception log says "A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond". I'm currently using the latest version of the library will switching to an older version be of any help. Also, thank you the library.

SmackyPappelroy, 2020-10-14:
@rjanak If you have issues with the network I don't think it is a good solution to close and open the connection in the same scan. 
Close the connection, and wait some time and then try to open it again. So the network has time to stabilize

rjanak, 2020-10-14:
@SmackyPappelroy thanks, I will try that. So basically this must be network related issue.

SmackyPappelroy, 2020-10-14:
That or Plc related! You can't make to many requesrs to the PLC at the same time

rjanak, 2020-10-14:
@SmackyPappelroy I'll keep that in mind. The thing is there are 5 pumps operating currently which are to monitored in real time for the amount of water delivered (in gallons). So im polling continously within a duration of 300 milliseconds. Ill increase the duration and try limiting the connections to the PLC and see. Thank You.

rjanak, 2020-10-14:
@SmackyPappelroy I see, but I have too many reads, and I have made separate connection for each pumps.
Also I have checked wireshark capture I see some packets with [TCP segment of a reassembled PDU] could that be indicative of any problem.

SmackyPappelroy, 2020-10-14:
@rjanak There is a parameter in the Plc settings how much computing power you want the Plc to use for communication. Default is 20 % I think. I usually boost this one to 50%. 
[image]

The Plc does not prioritize communication so it is good to boost this parameter. That might help

rjanak, 2020-10-14:
@SmackyPappelroy thank you.

scamille, 2020-10-17:
@mycroes What is your opinion on this? We really need to start a discussion here before doing anything. The real world experience with closing the connection manually is definitely helpful, but I still think part of it should be handled by the library itself, assuming we can clearly detect when a connection is in a invalid state.

qq357664624, 2023-11-29:
hi   我们发现在频繁循环 调用的过程中，S7NetPlus 执行.close() 方法并没有释放资源，会导致内存一直处于增长的状态，我感觉是个严重bug。  （频繁循环调用是指， 用open 打开一个不存在的地址， 判断对象是否存在）

gfoidl, 2023-11-29:
@qq357664624 please post comments in English, not everyone (at least me) can read what you have written. Thanks.

Here's a machine translation:

> We found that during the process of frequent loop calls, S7NetPlus executed the .close() method and did not release resources, which caused the memory to keep growing. I think it is a serious bug. (Frequent loop calls refer to using open to open a non-existent address to determine whether the object exists)