S7NetPlus/s7netplus #340: How to read all the data from DB at one time

FeigelChina, 2020-10-21:
If I try to read all the data from DB1 at one time, what should I do? DB1 has various types of value such as the real, word and so on. Every type also has various values. All the values are distributed on the DB1. Is there easy solution to read all the data from DB1?

757846197, 2020-10-21:
You can use Readclass to read all the data from DB1

757846197, 2020-10-21:
/// <summary>
        /// Reads all the bytes needed to fill a class in C#, starting from a certain address, and set all the properties values to the value that are read from the PLC. 
        /// This reads only properties, it doesn't read private variable or public variable without {get;set;} specified.
        /// </summary>
        /// <param name="sourceClass">Instance of the class that will store the values</param>       
        /// <param name="db">Index of the DB; es.: 1 is for DB1</param>
        /// <param name="startByteAdr">Start byte address. If you want to read DB1.DBW200, this is 200.</param>
        /// <returns>The number of read bytes</returns>

        public void readDB(object sourceClass, int db, int startByteAdr)
        {
            
            if (conectStatus == true)
            {
                try
                {
                    plc.ReadClass(sourceClass, db,startByteAdr);
                   
                }
                catch (Exception e)
                {
                    Logger.Instance.WriteException(e);

                }
            }

        }

757846197, 2020-10-21:
For example 👍 siemensPLC.readDB(db161, 161, 0);

FeigelChina, 2020-10-21:
So  I need to create a class generated from DB1 by myself. Is there a auto generation mapping the data types of DB1 to the class's properties?

FalcoGoodbody, 2020-10-21:
There is no auto generating mapping to create a class from a plc data block.
You need to create a class for your own or you can also read a list of data items.
Have you had a look in the wiki?

FeigelChina, 2020-10-27:
> There is no auto generating mapping to create a class from a plc data block.
> You need to create a class for your own or you can also read a list of data items.
> Have you had a look in the wik
as far as I know, some softwares can obtain the whole address books of PLC so that they can track all the values.

FalcoGoodbody, 2020-10-28:
But it is not implemented in this library.
As far as I know the optimized protocoll is not documentated and Siemens keep it like a secret. Companies which are building OPC-Servers like TANI or Softing are able to read also optimized data blocks but I do not know how they do it.
I think for this library it is maybe a long-time goal.

closed 2021-06-02 by mycroes