S7NetPlus/s7netplus #345: Unable to connect so SIMATIC S7 1500

Briasoft, 2020-10-27:
Using the following code
`                    _S71500 = new Plc(CpuType.S71500, this.txtIp.Text, Convert.ToInt16(this.txtRack.Text), Convert.ToInt16(this.txtSlot.Text));
                    _S71500.Open();
`
I receive back an exception: "couldn't establish a connection on <IPaddress>:102"
I've tryed even with a direct cable attached to the network module of the PLC to avoid network isues, but the error is still the same.
Any help woould be appreciate
Flavio

FalcoGoodbody, 2020-10-28:
Did you turn offed the access level an turned on PUT/GET Access as described in the wiki?

Briasoft, 2020-10-29:
Thanks for reply, the PLC has been set up from another firm, they told me that PUT and GET are enabled, I'll ask them about the access level. Could the SECOMEA device create connection problems?

FalcoGoodbody, 2020-10-29:
What is a 

> SECOMEA

device?

Briasoft, 2020-10-29:
Is a remote connection device that the contructor uses to gain remote access to the plant, and il fisically bridged to PLC ethernet unit. I'm guessing that there is a problem between the ethernet module and the PLC unit.

FalcoGoodbody, 2020-10-29:
so it´s like a switch or a router??
possible that you need to configure this correctly

Briasoft, 2020-10-29:
I've tryed even with this device disconnected, plugging directly into the ethernet module but the problem still persists. Now the error message is changed to Comunication interrupted at transport level

scamille, 2020-10-29:
S7 protocol uses TCP protocol, and by default Port 102.

You definitely need to ensure that you can reach the PLC through those means. Having a successful ping would be the first test, but you also need to ensure that Port 102 is not blocked by your firewall.

Briasoft, 2020-10-29:
The PLC answers at ping, and using telnet I can open a session on port 102, bytheway when I call the open function it seems that somehow the PLC is firewalled, even with a direct cable connected to the ethernet module. The inner error of the exception says wrong IP

FalcoGoodbody, 2020-10-30:
What about the access level? Coud you check if it is turned off?

Briasoft, 2020-10-30:
I've asked, is set to full access, complete control (Answer from the manufacturer)

Briasoft, 2020-11-03:
I've found a similar issue [#128 ] that seems to be related to mine, but I'm using a SIMATIC S7 1500 CM. There could be a similas problem on this one?

Briasoft, 2020-11-06:
The problem has been solved, there was a network problem on the managed switch that closes the port 102

closed 2020-11-06 by Briasoft

asaad5556, 2024-10-02:
Hi all,
My problem is that I am unable to establish a connection between my Weintek display and a PLC 1500. The ‘Connection Mechanisms’ option is grayed out and cannot be activated.

Do you have any suggestions on how to resolve this connection issue?

asaad5556, 2024-10-02:
[image]