S7NetPlus/s7netplus #346: OpenAsync with timeout function

barnabaskisstdk, 2020-10-28:
Hi,

I would like to ask whether there is some possibility to set some timeout under OpenSync. I tested the method and approx. after 20 seconds I got error message when the PLC is not ready. How can I parametized it?, or would you be so kind and build in this parameter in the OpenAsync method? i.e OpenAsync(int timeoutInMillisecond)

thx in advance.

FalcoGoodbody, 2020-10-28:
@scamille, @mycroes any chance there?

scamille, 2020-10-28:
Cancelling a connection attempt is a bit of a tricky thing, as far as I understand it.
The NET functions for example don't even offer CancellationToken support, see https://docs.microsoft.com/en-us/dotnet/api/system.net.sockets.tcpclient.connectasync?view=netcore-3.1 before net5.

There are tricks like https://stackoverflow.com/questions/14635422/how-to-stop-the-socket-connect-function-if-it-stucks which might allow cancelling it, but I am not sure how good that works in practice on different systems and .NET versions.

In any case, right now this is not supported by S7NetPlus.

barnabaskisstdk, 2020-10-29:
Hi,

Thanks for your answer.

I can stop the OpenAsync, if I use timeout under 20 sec. When I use for example 30000 milliseconds, my timeout priority will be lower than the OpenAsync "own" priority, and the OpenAsync method will be stopped after around 20 sec.

                if (await Task.WhenAny(SiemensPLC.OpenAsync(), Task.Delay(10000)) == SiemensPLC.OpenAsync())
                {
                    // task completed within timeout
                }
                else
                {
                    // timeout logic
                    sw.Stop();
                }

scamille, 2020-10-29:
You need to increase your operating systems default socket connection timeout value to get a longer timeout.

barnabaskisstdk, 2020-10-29:
I tried many ways but without success. I have Win 10 Pro, and set TcpInitialRTT and TcpMaxConnectRetransmissions parameters in the system registry.
Default values: 3 and 2 which give the 21 sec (3+6+12)

Would you be so kind and help me which another way should I try it?

Thx in advance.

scamille, 2020-10-29:
I don't have any particular knowledge about this.
Google gave me for example this stackoverflow answer: https://stackoverflow.com/questions/22341885/increase-timeout-for-system-net-sockets-socket-connect

barnabaskisstdk, 2020-10-29:
Hi,
 I solved the problem.
I set the netsh interface tcp set global initialRto=3000
The Max SYN Retransmissions = 4

The default op. system timeout is 93 sec (3+6+12+24+48)

Thx for your initial help.

Process:

CMD in admin mode
netsh interface tcp set global <for help>
netsh interface tcp show global

TCP Global Parameters
                ----------------------------------------------
                Receive - Side Scaling State            : enabled
                Receive Window Auto-Tuning Level        : normal
                Add - On Congestion Control Provider    : default
                ECN Capability                          : disabled
                RFC 1323 Timestamps                     : disabled
                Initial RTO                             : 3000
                Receive Segment Coalescing State        : enabled
                Non Sack Rtt Resiliency                 : disabled
                Max SYN Retransmissions                 : 4
                Fast Open                               : enabled
                Fast Open Fallback                      : enabled
                HyStart                                 : enabled
                Pacing Profile                          : off

mycroes, 2021-06-21:
Closing this issue, as I guess this is an edge-case that doesn't really belong to S7netPlus. It's possible to connect to something at the other side of the world within a few milliseconds, so it baffles me you need more than 20 seconds to connect to a PLC. Either way you found yourself a workaround, so you should be fine. Do note that v0.13.0 introduces a default timeout of 10 seconds in S7NetPlus that you can still set to a custom value.