S7NetPlus/s7netplus #350: value conversion

FeigelChina, 2020-11-20:
how to read the address like DB2119.DD32?
From the Wiki, I don't find the related value conversion about "DD".
When I read the source code, it does not mention the situation in case of "DD". Look at the attached screenshort.

[image]

FalcoGoodbody, 2020-11-27:
Hello @FeigelChina ,

what datatype in S7 should DD be?
A double word is DBD.

Can you show an example of the data you want to read?

FeigelChina, 2020-11-27:
Hello @FalcoGoodbody  
The level 1 co-workers give me the wrong data address. They left out a letter "B".  There is no datatype in S7 should DD be. 
Thank you very much!

closed 2021-06-02 by mycroes