S7NetPlus/s7netplus #354: Can't read PLC DB Tag of "Array[0..2] of myDataType"

tadej-ihs, 2020-12-14:
DB in TIA PORTAL
[image]

My Model object named TestDB3
[image]

Json result of ReadClass function
[image]

Hi!

I'm trying to read values from DataBlock into Asp.Net Core 3.1 class. I'm using ReadClass\<T\> function to get data, where T is of type TestDB3.
It works fine reading from property named "one" to "twelve". Then at "thirteen" there is reading mismatch. At "fourteen" there is again everything as expected.

As a mismatch I have in mind that only "thirteen[0]" gets correct values for "c" and "d" property, then at "thirteen[1]" both properties "c" and "d" are left on default value regardles of being set to "true" on PLC DB side,

I experience that I'm limited to tranfer only basic data types, arrays of basic data type and single custom data type. Can I transfer array of custom data type like i show with property "thirteen"?

If i'm doing something wrong, correct me please :) 

Thanks! 

Tadej

scamille, 2020-12-14:
Please try out using this as a replacement for you thirteen class:
```
public class thirteen
{
  public bool c {get;set;}
  public bool d {get;set;}
  public byte Dummy {get;set;}
}
```

My guess is that calculation of adresses is not correct in S7NetPlus, and it does not align correctly to the even adresses.

tadej-ihs, 2020-12-15:
Hi! 

Your quess is correct. It now works fine. 

It means that "array[0..x] of bool" is inclined to mistakes correlated to addresses or better say "PLC offsets". With included " dummy" byte secound array item reads from right values.

Thanks a lot!

closed 2020-12-15 by tadej-ihs

scamille, 2020-12-15:
As far as I understand it, UDTs are always of an even size. So if you have a odd number of bytes in it (like you have with your 2 bools, which make up 1 byte), the size of the UDT is still 2 bytes and not 1.

Automatic calculation of this is probably something that should be added to S7NetPlus as a improvement. When something like this is implemented, the solution with the Dummy byte should still not break in any way since you are now already at a even size.