S7NetPlus/s7netplus #355: Xamarin & .NET Standard 2.0

Pallo100, 2021-01-04:
**Good Year!!!** Sorry for the inconvenience, I would like to create an APP to send commands to my S7-1215 PLC. If I create a UWP everything works correctly but if I create an APP with Xamarin .NET Standard 2.0 or 2.1 I get the following error:

{System.ArgumentException: Value does not fall within the expected range.
  at System.Net.Sockets.Socket.SetSocketOption (System.Net.Sockets.SocketOptionLevel optionLevel, System.Net.Sockets.SocketOptionName optionName, System.Int32 optionValue) [0x00023] in /Users/builder/jenkins/workspace/archive-mono/2020-02/android/release/mcs/class/System/System.Net.Sockets/Socket.cs:2709 
  at System.Net.Sockets.TcpClient.set_SendTimeout (System.Int32 value) [0x00000] in /Users/builder/jenkins/workspace/archive-mono/2020-02/android/release/mcs/class/referencesource/System/net/System/Net/Sockets/TCPClient.cs:572 
  at S7.Net.Plc.ConfigureConnection () [0x00026] in <02b123a5064f4c48af7351babd69d8d5>:0 
  at S7.Net.Plc.ConnectAsync () [0x0001c] in <02b123a5064f4c48af7351babd69d8d5>:0 
  at S7.Net.Plc.OpenAsync () [0x0007f] in <02b123a5064f4c48af7351babd69d8d5>:0 

do you have any ideas or suggestions?

Thanks!!!!
Ciao
Paolo

mycroes, 2021-01-04:
Hi @Pallo100,

I believe this issue is already fixed, but not yet released. Unfortunately I tend to be short on time, as such the release carrying those features has been missing.

In more detail with regards to the issue, originally I used `System.Threading.Timeout.Infinite` for read and write timeouts, but that apparently breaks on Mono. It's been changed to a plain `0` now, which should work without issues.

Anyway, I'll try to push out a new release (but I believe something in the CI pipeline was broken) to have this fixed in the package.

Regards,
Michael

mycroes, 2021-01-04:
Just released [v0.8.0](https://github.com/S7NetPlus/s7netplus/releases/tag/v0.8.0). Should be available from nuget.org within an hour or so. Closing this issue, feel free to reopen if it persists with 0.8.0.

closed 2021-01-04 by mycroes

Pallo100, 2021-01-04:
Thanks!!! this fix work for me. IOS 12.5 and Android 9