S7NetPlus/s7netplus #357: OverflowException in 0.8.1

SmackyPappelroy, 2021-01-06:
I am developing a new application and am getting an overflow exception after a while when trying to read a tag. This application in developing stage so it might be an error on my part. But I don't get why. Any ideas?
[image]

SmackyPappelroy, 2021-01-07:
I was able to resolve this issue myself by including a thread-lock, due to the fact that I read values on a seperate thread as well. This was my fault

closed 2021-01-07 by SmackyPappelroy

mycroes, 2021-01-07:
Great. One tip, use the async API and use proper async constructs, that'll make your life a lot easier in the long run. Still requires synchronization in that case, but you can create an exclusive task queue for async operations in like 20 lines of code. Are you creating something as a hobby project, or professionally? I'd be willing to invest some time in a mobile sample application, so if you want any help, let me know.

SmackyPappelroy, 2021-01-07:
I am developing an application for analyzing "Cleaning In Place"-systems in food processing industries. At the moment it is just on a hobby level. I need to get it to work properly before I implement this on a professional scale. I am a beginner to intermediate programmer so I am not that good at it ;)
I have refrained from using the async API, due to the fact that I think it is like a "virus" once you start to use async programming. Everything need to be async and that creates problems for my programming. But I am very much interested in help with a sample application. As I mentioned my programming skill is probably a tenth of yours. Best regards! @mycroes

mycroes, 2021-02-05:
Hi @SmackyPappelroy, sorry for my (very) late response. I'm currently working on a graphical application to read data from S7 and Modbus connections. Although I'm currently developing it around [Sally7](/mycroes/Sally7) it could work with S7NetPlus just as well. I hope to have a very barebones prototype soon, which should then at least support PLC connection configuration and reading variables. You can find the project at [mycroes/PlcMonitor](/mycroes/PlcMonitor), as of this moment it supports the following:
- Setting up a PLC connection using S7 or Modbus protocol
- Testing the connection during connection setup
- Saving project to a JSON file
- Loading project from a JSON file

The UI isn't great (and that's an understatement), there's some code quality issues, the stored data isn't properly loaded and there are more issues, but it's a work in progress. I can't even promise it actually compiles right now (but I think it does 😄 ), there's no verification except my manual checks. Still, feel free to check it out, in a week it could actually be in some working state.