S7NetPlus/s7netplus #358: Is is possible to write values to the input area to the PLC?

hydudajan, 2021-01-08:
Hey guys,

First of all, I have to say that this project is really great.
Now to my question. Is it possible to write a value in the input area of the PLC?

I would like to use my PLC to test my logic and build a simulation for it with C #. For a conveyor system, for example. In this simulated system there would also be sensors whose values I would like to use as inputs in the PLC as usual. 
So are we able to write values inside the input memory to the PLC?

Do you know if it is possible to realize that? Or do I have to rethink my structure?

FalcoGoodbody, 2021-01-11:
Hey @hydudajan,

the input-area of a plc can not be written.
In generall there are sensor devices or something connected to the physical inputs.
If you try to set the input it will be overwritten by the plc with the actuall state of the physical input within the next cycle (< 10ms).
One can only use a so called force-table within the IDE of SIEMENS to force the inputs to specific logical value.
But one should be very very carefull when using this.
And is not accessible with this library.

hydudajan, 2021-01-11:
I am working in automation industry here in Germany. So i am used to these PLC'S and know about the force table. But i thought about setting the inputs, like my normal input-modules would. 
But when we are not able to write to inputs, i need to rethink my solution.

Thank You

closed 2021-01-11 by hydudajan