S7NetPlus/s7netplus #360: Access PLC in other C# project pages- S7netplus

midelav, 2021-01-11:
Dear all,
I working on a UWP C# application to communicate with S7-300 PLC, and the driver is S7netpuls v0.8.1
If I set a PLC instance on main page and try to write values in PLC, every thing works fine. But I need to Write PLC values from other pages. The issue is that if I define a new plc instance in other pages and try to run this example code: SamplePlc.Write("DB5.DBX7.6", true);  because of an existing connection, it is impossible to establish a new  connection and write the example code.
How can I access the plc in other project pages to write plc tags?

hydudajan, 2021-01-11:
In my opinion you should create a Static "PLC Manager" class. This could contain a public static PLC property, which can then be accessed from any other class. Establishing a connection could also be done in this PLCManager class.

```
using S7.Net;

	public static class PlcManager
	{
		private static Plc _plc;
		
		public static Plc Plc => _plc;
		
		static PlcManager()
		{
			_plc = new Plc(CpuType.S7300, "192.168.178.1", 0, 2);
		}
	}

	public class Test
	{
		public void ReadDataFromPlc()
		{
			var plc = PlcManager.Plc;

			if (plc != null)
			{
				plc.Open();

				plc.ReadBytes(DataType.Input, 0, 0, 10);
			
				plc.Close();
			}
		}
	}
```

mycroes, 2021-02-08:
Having a PlcManger is a good idea, however there's no reason for the PlcManager to be static. You could just as well have this as a single instance in your application and apply DI/IoC to inject it into the services where you need it. Either way, I'm closing this issue because it's not S7NetPlus related.

closed 2021-02-08 by mycroes