S7NetPlus/s7netplus #364: read plc DB error

stonellearning, 2021-01-25:
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.NetworkStream'.
   at System.Net.Sockets.NetworkStream.ReadAsync(Byte[] buffer, Int32 offset, Int32 size, CancellationToken cancellationToken)
   at S7.Net.StreamExtensions.ReadExactAsync(Stream stream, Byte[] buffer, Int32 offset, Int32 count)
   at S7.Net.TPKT.ReadAsync(Stream stream)
   at S7.Net.COTP.TPDU.ReadAsync(Stream stream)
   at S7.Net.COTP.TSDU.ReadAsync(Stream stream)
   at S7.Net.Plc.ReadBytesWithSingleRequestAsync(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   at S7.Net.Plc.ReadBytesAsync(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   at CoreLibrary.PLC.ReadCore.PlcReadCore.ReadBytesAsync(Int32 db, Int32 startByteAdr, Int32 count) in

stonellearning, 2021-01-25:
System.IO.IOException: Unable to read data from the transport connection: 由于线程退出或应用程序请求，已中止 I/O 操作。.
 ---> System.Net.Sockets.SocketException (995): 由于线程退出或应用程序请求，已中止 I/O 操作。
   --- End of inner exception stack trace ---
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.GetResult(Int16 token)
   at System.Threading.Tasks.ValueTask`1.ValueTaskSourceAsTask.<>c.<.cctor>b__4_0(Object state)
--- End of stack trace from previous location where exception was thrown ---
   at S7.Net.StreamExtensions.ReadExactAsync(Stream stream, Byte[] buffer, Int32 offset, Int32 count)
   at S7.Net.TPKT.ReadAsync(Stream stream)
   at S7.Net.COTP.TPDU.ReadAsync(Stream stream)
   at S7.Net.COTP.TSDU.ReadAsync(Stream stream)
   at S7.Net.Plc.ReadBytesWithSingleRequestAsync(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   at S7.Net.Plc.ReadBytesAsync(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   at CoreLibrary.PLC.ReadCore.PlcReadCore.ReadBytesAsync(Int32 db, Int32 startByteAdr, Int32 count) in

stonellearning, 2021-01-25:
System.OverflowException: Arithmetic operation resulted in an overflow.
   at S7.Net.TPKT.ReadAsync(Stream stream)
   at S7.Net.COTP.TPDU.ReadAsync(Stream stream)
   at S7.Net.COTP.TSDU.ReadAsync(Stream stream)
   at S7.Net.Plc.ReadBytesWithSingleRequestAsync(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   at S7.Net.Plc.ReadBytesAsync(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   at CoreLibrary.PLC.ReadCore.PlcReadCore.ReadBytesAsync(Int32 db, Int32 startByteAdr, Int32 count) in

stonellearning, 2021-01-25:
How can I solve it?

stonellearning, 2021-01-26:
2020-11-18 09:06:34,893 [26] ERROR: read plc DB:32 error
S7.Net.TPKTInvalidException: TPKT is incomplete / invalid
   at S7.Net.TPKT.ReadAsync(Stream stream)
   at S7.Net.COTP.TPDU.ReadAsync(Stream stream)
   at S7.Net.COTP.TSDU.ReadAsync(Stream stream)
   at S7.Net.Plc.ReadBytesWithSingleRequestAsync(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   at S7.Net.Plc.ReadBytesAsync(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   at CoreLibrary.PLC.PlcReadCore.ReadBytesAsync(Int32 db, Int32 startByteAdr, Int32 count) in E:\WorkProject\SmartGuy\Longli\SourceCode\ApiCode\WinService\CoreLibrary\PLC\PlcReadCore.cs:line 272

mycroes, 2021-01-26:
There recently was someone else that ran into the `OverflowException`, I believe the cause was related to multithreaded access to the PLC instance in that case. Also, you're just posting a bunch of stack traces, that isn't really helpful. Maybe you could share an excerpt of the code you're using? Also, I'd recommend you start with something small like a console program to read one variable only. Your issue seems to indicate improper use of the library (like multithreaded access), not an issue within the library.

stonellearning, 2021-01-27:
2021-01-20 16:01:47,188 [5] INFO : read plc DB:51 error
System.OverflowException: Arithmetic operation resulted in an overflow.
   at S7.Net.TPKT.ReadAsync(Stream stream)
   at S7.Net.COTP.TPDU.ReadAsync(Stream stream)
   at S7.Net.COTP.TSDU.ReadAsync(Stream stream)
   at S7.Net.Plc.ReadBytesWithSingleRequestAsync(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   at S7.Net.Plc.ReadBytesAsync(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   at CoreLibrary.PLC.ReadCore.PlcReadCore.ReadBytesAsync(Int32 db, Int32 startByteAdr, Int32 count) in 
[image]

stonellearning, 2021-01-27:
public async Task<byte[]> ReadBytesAsync(int db, int startByteAdr, int count)
        {
            try
            {
                using (plc)
                {
                    if (!plc.IsConnected)
                        plc.Open();

                    return await plc.ReadBytesAsync(DataType.DataBlock, db, startByteAdr, count);
                }
            }
            catch (Exception ex)
            {
                //_logService.Debug($"read plc DB:{db} error", ex);
                _logService.Information($"read plc DB:{db} error", ex);

                return default;
            }
            finally
            {
                Close();
            }

        }

mycroes, 2021-01-27:
You shouldn't be having a `using` block there, would start by removing that.

stonellearning, 2021-01-28:
Like "return await plc.ReadBytesAsync( db, startByteAdr, count);"
but  datatype parameters are required in methods

mycroes, 2021-01-28:
No, I'm talking about the `using` block. Are you a C# developer, or are you just using C# to try out S7NetPlus?

stonellearning, 2021-01-28:
I am a C# beginner，I need to use C# to read the value of DB block with PLC DB block

stonellearning, 2021-01-28:
[image]
I don’t  add the using block，as shown in the picture

mycroes, 2021-01-28:
Your original code is this (would be nice if you'd also had just pasted the code properly instead of a screenshot):
```csharp
using (plc)
{
    if (!plc.IsConnected)
        plc.Open();

    return await plc.ReadBytesAsync(DataType.DataBlock, db, startByteAdr, count);
}
```

I only took the part within the `try` block, because that's the relevant part. The code from above should become this:
```csharp
if (!plc.IsConnected)
    plc.Open();

return await plc.ReadBytesAsync(DataType.DataBlock, db, startByteAdr, count);
```

I can't promise it'll work then, because this is just a small excerpt from the code, but this is definitely wrong and will cause at least one of the exceptions you posted.

In all honesty, this is not the place to ask about general C# programming and that's what you're doing. I can only suggest you get to know C# first and then try to apply that knowledge to external libraries.

closed 2021-01-28 by mycroes