S7NetPlus/s7netplus #369: Why so  there's  so huge difference between assembled dll and dll which was assembled by me.

Wobeam, 2021-02-25:
Hello  everyone,
In older projects I used  v.0.1.9  for them it was taken already assembled dll which had location  _s7netplus-0.1.9/nuget-pack/lib/net45/s7net.dll._
It worked perfectly.
Today, just out of curiosity I've tried to add to solution S7.Net project and assembled dll by yourself.

There's huge difference. This works but it is so faulty comparing dll which was assembled by developers.

I believe I did something wrong. But what?

Any comments would be appreciated.

mycroes, 2021-02-26:
I have no idea what differences you're talking about, please explain...

Wobeam, 2021-02-26:
Ok,

- if take ready to use dll here  _s7netplus-0.1.9/nuget-pack/lib/net45/s7net.dll._  the data exchange in App is stable.
- if assemble dll by yuorself using s7netplus-0.1.9  S7.Net Class library - this dll shows very unstable results. The data exchange breaks  a few tims per minute.

I have no clue what could be a difference between these DLLs.

mycroes, 2021-02-27:
Not that I know of. There might be differences due to debug or release configuration and probably you're using a newer compiler than what 0.1.9 was built with, although I doubt that would matter much.

However, 0.1.9 is three years old. It has a few defects that have been fixed and overall reliability should be better, so please just use a recent release instead.

Wobeam, 2021-02-27:
@mycroes,
followed your advice and ended up with dll  v7.0. 
the version 8.0 has thrown a fail as per picture attached.
What could be a reason between 0.1.8 - 7.0 that cause a fail?

thanks.[
[image]
](url)

mycroes, 2021-02-27:
The latest is 0.8.1, and what you're showing is not a failure (did you even read the message?) I can imagine however that you're using IsAvailable to check for the connection state, which you shouldn't be doing anyway. No property check can ensure that the next call will succeed, so just make the call and handle exceptions appropriately. IsAvailable was breaking the connection for a long time, I'm unsure about it's current state.

Wobeam, 2021-02-28:
@mycroes,
this message I read and it saud me nothing useful 

> The function evaluation requires all threads to run.

I'm not using IsAvailable at all. 

With the simplest connection code below I went through all available Dlls beginning with 0.1.8 and ending with 7.0 . They all are workable. The data exchange takes place with all of them.

```
            try
            {
                plc = new Plc(CpuType.S71500, "100.1.1.70", 0, 0);
              
                plc.Open();
              
            }

            catch (Exception ex)
            {
            
                MessageBox.Show(ex.Message);

            }

```
However, beginning with v.8.0 the piсture is changing to the opposite,
There are not data exchange and any exceptions.
The picture while debugging I had shown in previous post.

mycroes, 2021-02-28:
I guess you're making these calls from the ui thread, there might be an issue there with regards to how the sync api now just invokes the async api. Could you change the method to be async and use the OpenAsync method? If that works, I have an idea about what to look for.

Wobeam, 2021-02-28:
@mycroes,
if I got you correctly

I did as follows:
 ```
private async void Connect()
        {

            try
            {

                plc = new Plc(CpuType.S71500, "100.1.1.70", 0, 0);

                await plc.OpenAsync();

            }

            catch (Exception ex)
            {

                MessageBox.Show(ex.Message);

            }

        }

```
However, the issue still persits.

mycroes, 2021-02-28:
That's right. Could you try to connect using a command line application to see if the issue relates to the fact that this is on the ui thread at all?

scamille, 2021-02-28:
Hmm so you suspect that the different synchronization context in a gui application might break the sync calls?
We probably never tested that, since unit tests behave like command line applications.

 Unfortunately I don't have that much experience/knowledge about this. Shouldn't ensuring that there is ConfigureAwait(false) used everywhere internally fix the problem?

Wobeam, 2021-02-28:
@mycroes,
again, if I got you correctly,

**this** picture below has been made with v7.0.

[image]

**this** picture below has been made with v 8.0

[image]

Is theree any key difference between 7.0 and 8.0?

scamille, 2021-02-28:
S7 Real maps to .NET float / System.Single, while LReal maps to .NET double / System.Double

Wobeam, 2021-03-02:
Ok, thanks.
what about integers? 
Plc side     .Net side
Int              ?
DInt           ?

mycroes, 2021-03-02:
Types of matching size, so:
Int => `System.Int16` or its alias `short`,
DInt => `System.Int32` or its alias `int`.

Wobeam, 2021-03-03:
Hello,
at early version there have been done an opportunity to read ErroreCode
as follows:
`if (errCode != ErrorCode.NoError) throw new Exception(plc.LastErrorString);`
It was quite helpful.
What about the lattest ones? They don't have that. So how to handle with such kind of errors?

thanks.

mycroes, 2021-03-03:
That's been changed to throw relevant exceptions internally. Returning error codes and LastError properties are more C-like, they're not considered good practice in C#. Changes have been made to make the library more conformant with what is common practice in the C# world.

Mallowan, 2021-03-03:
@Wobeam, I would note that reasonable questions you are asking here. I had same doubts.
Reffering  to the fist your post, may I ask how you appreciate this library. Is there something running at industrial area based on this lib and made by you?

Wobeam, 2021-03-04:
Hi,
you know I used an early version a couple of years ago. This works decently up to now despite of I hvave been the world's worst C# programmer at that time. Today I am also bad but not so much as two years ago.  ) 
Now there's a task to make something else. Much water has gone under the bridge an we have v8.1 here available.
I can not tell anything about reliability of the lattest one.
The only thing I can tell you that simple replacement of 0.1.9 with 8.1 caused some fails. Currently I'm trying to solve them. Thanks to @mycroes and @scamille the things are moving.

Hope owners or someone experienced find 5 minutes to help me out. )
At the moment I am highly interested in how to address to plc instance from a number of viewmodels and where place plc.Open() method within such a program structure.
Of course, I learned it long ago how to create instances of classes and then address to their fields. 
But this is communication library, and I really want to know how to do that properly in order to get **flawless** code.
I'm having some my own good practices which I used. However could someone put here a piece of code explaining my question.

thanks.

mycroes, 2021-03-04:
@Wobeam I'm actually working on a [UI](https://github.com/mycroes/PlcMonitor) for interacting with PLC's, the part that manages the PLC connection is at [src/UI/Models/Plcs/Plc.cs](https://github.com/mycroes/PlcMonitor/blob/main/src/UI/Models/Plcs/Plc.cs).

It's the first time I'm managing concurrency based on observables, but it works reasonably well. There's probably a 100 approaches you can take to managing concurrency and keeping track of PLC instances, but honestly I think you should be looking for such general advice elsewhere. It's also impossible for me to decide what the best fit for you would be, because I don't write your application...

scamille, 2021-03-04:
I had a quick look at the ConfigureAwait settings, and I do think we are all covered.

@Wobeam without an actual error message / stack trace it is hard to understand what your actual problem is. Your picture is just showing some local variables while you are debugging, but nothing about the actual problem you encountered.

Wobeam, 2021-03-04:
@scamille, 
seems I solved issues with types convertation. I left App running and organized constant data exchange for a night. so far there have not been any hangs. So I'm going to encrease traffic load.
fingers crossed.)

mycroes, 2021-03-07:
Closing this issue because it seems fixed now, feel free to continue discussion or open new issues if other problems arise.

closed 2021-03-07 by mycroes