S7NetPlus/s7netplus #370: Can't migrate to lattest version. Need an explanation.

bolt-b, 2021-03-07:
Hello guys,
untill now we had 0.1.8 version implemented in the project. And it worked fairly fine. 
It is time to make another application and we came accross with some challenges.
Early versions allowed to read multiple DB as per example code below.

```
ReadMethod()
{   
            DB1 db1 = new DB1();
            plc.ReadClass(db1, 1); // reading DB1
          
            value0 = db1.varReal1;
            value1 = db1.varInt1;
            value2 = db1.varReal2;
            value3 = db1.varInt2;
            value4 = db1.varReal3;
            value5 = db1.varInt3;
            value6 = db1.varReal4;
            value7 = db1.varInt4;
            value8 = db1.varReal5;
            value9 = db1.varInt5;
            value10 = db1.varReal6;

            //DB2 db2 = new DB2();
            //plc.ReadClass(db2, 2); // reading DB2

            //db2_value1 = db2.M_0;
            //db2_value2 = db2.M_1;
            //db2_value3 = db2.M_2;
            //db2_value4 = db2.M_3;
            //db2_value5 = db2.M_4;
            //db2_value6 = db2.M_5;
            //db2_value7 = db2.M_6;
            //db2_value8 = db2.M_7;
            //db2_value9 = db2.M_8;
            //db2_value10 = db2.M_9;
            //db2_value11 = db2.M_10;
}
```

Now, it seems not an workable way to read multiple data blocks. Or?
Could someone help us to migrate to lattest version? Basically the question is about  a way of reading multiple data blocks.

Thanks in advance!

mycroes, 2021-03-07:
First off, you're missing any technical detail about what's going wrong. Do you get exceptions? Do you get incorrect data? Does your cat explode on read of DB2? You're saying that the question basically is about multiple reads, but I don't see any actual question.

Secondly, you might be running into the same issues as in #369. To start, check if your real variables are mapped to C# float and not double, because double is the wrong type for real and no longer works as such.

bolt-b, 2021-03-07:
We read a dozen of classes by a way as in the first post.  Before migration to 8.1 I left only two classes (for the sake of simplicity) to be read and replaced all doubles with floats, thanks to #369 that is helpful. 
There are not any exceptions there.
Based on my observations I would say that library lost its reliability. At least in my particular case.
To be more concrete, it may connect to 1500 series plc, it may not connect.
It may read a few seconds both data blocks properly, then UI freezes for some unknown reasons.
I spent hours to catch  a dependency or a reason or so, but to no avail.
Once I attached back to the project  the lib of version 0.1.8  everything becomes OK. A dozen of pretty heavy DB are being read, there are not any freezes. The data are plausible.
In other words it becomes relatively reliable thing.

If you have any further quesion please ask.

mycroes, 2021-03-07:
As stated in my previous reply, I still don't know if your cat explodes. You're not detailing anything yet. There's no class layout, there's no code that fails, there's no detail of any exception whatsoever.

The fact that your UI freezes could be due to issues surrounding the synchronization context (even though there shouldn't be any) or the fact that you're performing calls on the UI thread that apparently don't complete (also shouldn't be happening, but perhaps a possibility).

There have been changes to how data is being received in the sense that previously if not all data was available in a single read from the `NetworkStream` then the code would just throw, while nowadays it tries to read until all data is received. I personally never noticed receiving less data than the entire message (however, that's not through S7NetPlus), but it's a change from old behavior...

Unfortunately as long as your description is vague I'm not able to provide a solid answer either, even moreso I don't have anything to go by to search for possible bugs.

bolt-b, 2021-03-08:
@mycroes, hi
this is communication part of the code
```
 try
            {               
                plc = new Plc(CpuType.S71500, "20.20.20.2", 0, 0);

                //errCode = plc.Open(); // back when 0.1.8 utilization

                if (!plc.IsConnected)
                plc.Open();
                
                //if (!plc.IsAvailable) throw new Exception("Plc not responding to ping"); // back when 0.1.8 utilization

                //if (errCode != ErrorCode.NoError) throw new Exception(plc.LastErrorString); //back when 0.1.8 utilization                      
            }

         
```

Can you explain what you meant by _cat explode_?
Also how to catch any exceptions when there are not any fails while debugging and when UI simple freezes after a while?
I can judge about library behaviour only by taking as a base the observations performed while application launching and starting.

Wobeam, 2021-03-08:
hello everyone,
@bolt-b  I am from #369 you have recenly mentioned here.
let me put my two cents in your discussion.
I wrote that I used 0.1.9, however as it turned out in fact I used 0.1.8!
Because, devs confused at that time and had put to 0.1.9 nuget the version 0.1.8.
It may be easily checked by means of comparison both dll sizes. So  I couldn't  use 0.1.9 as a connected project. That was a reason.
Well, nevermind.

Thus I'm trying to say that 0.1.8 is still OK for multiple class reading by a method you pointed out above.
So, I have similar issues which are observed as UI freezes. Or when App had started at last and performing reading when you tries to write a bit the UI freezes anyway. 
The 0.1.8 simple **lacked** of all mentioned disadvantages.

FalcoGoodbody, 2021-03-08:
@bolt-b may it possible that you use the wrong rack number for your S7-1500? I checked my code and everywhere I connect to a 1500, the rack number is 1 and not 0 as it is in the hardware configuration in TIA-Portal.
Can you check that please?

mycroes, 2021-03-08:
@bolt-b With _cat explode_ I simply mean the you're not providing anything to go by (you know, cats shouldn't explode when you connect to a PLC). Even now you're just pasting part of your code, but then you complain about UI freezing while there's no UI interaction in your code sample nor is there any UI interaction in S7NetPlus. Also, you're saying your UI freezes _after a while_, so what actually happens _after a while_?

Have you considered using the async API and properly integrating that into your UI? It seems that almost everyone that's running into issues is using the sync API, so I guess I might be better off by removing it entirely.

Wobeam, 2021-03-08:
@FalcoGoodbody hello and thanks for the input, 
 Upon slot,
 it doesn't matter much as it was a couple years ago and now. Seems it doesn't affect communication at all.
Also take a look at intellisense below.
[image]
Intellisense recommends both to be zeros. You can safely replace 1 with 0 and your application willbe still working.

Upon IsAvailable,
IsAvailable for 0.8.1 it is commented.
for 0.1.8 it is a part of code and works fine.

Bolt

mycroes, 2021-03-08:
It actually does make a difference for S7 1500:
[image]

I'm not sure what ID's the PLC accepts, I always use `03:01` if I want an unconfigured TSAP, which would be the same as passing `0` for rack and `1` for slot.

bolt-b, 2021-03-08:
Guys,
having in TIA portal : rack 0 and slot 1 did some tests.
-playings with rack caused always exceptions as "couldn't establish conneextion with 20.20.20.2". So, to play with rack is prohibited ). It must be exactly as in TIA.
@Wobeam you are right, playings with slots don't affect at all.

@mycroes,

> Also, you're saying your UI freezes after a while, so what actually happens after a while?

I have  two loops in plc each produces i+0.1, and i+1 at its output.
The application just gets these outputs values at TextBlocks. 
So,
-I can see "living numbers" for 24/7 when it is used 0.1.8. Really reliable stuff!
-I can see "living numbers" a few seconds when it is used 0.8.1, then "living numbers" just freeze.

mycroes, 2021-03-08:
@bolt-b I'm not sure how I should phrase this, because it doesn't seem to land: what is it you're doing, technically? Again there's a complete lack of technical details, which probably is the only thing that matters in this case.

- What is your target platform?
- What framework are you targeting?
- Are you using a timer to read data?
- If yes, what type of timer are you using?
- What thread is accessing the PLC instance?
- Do you have multiple threads accessing the PLC instance?
- How do you update your UI?
- Do you have exception handling in place around all PLC interactions?
- Does it always freeze after a few seconds (or what is the maximum time you've seen it running)?

Also, not a technical issue, but could you describe how experienced you are as a C# programmer?

I understand there might be some frustration about the library not behaving as it previously has, but there's still nothing that points at an issue with the library. For all I know the library could now be faster (or slower) than before and that causes issues with your application. There have been a few changes that delegate some sync methods to their async implementation and wait for the result, however there don't appear to be any issues related to those changes.

If you can upload the project I'd be happy to take a look, see if I can reproduce the issue.

bolt-b, 2021-03-08:
> What is your target platform?

it is 4.5.2 (btw, 0.8.1 is also targeting 4.5.2 by default)

> What framework are you targeting?

WPF App (.Net Framework)

> Are you using a timer to read data?

Yes I am. It's System Timer.

> What thread is accessing the PLC instance?

I created a class library and connected it to solution. This class library has a class where I make a connection to plc.

> Do you have multiple threads accessing the PLC instance?

This is quite big application, so my answer  -yes. Yet while troubleshooting others are disabled.

> How do you update your UI?

An event subscribed to a timer, then it is being broadcasted to a method which is located in VM (used mvvm pattern).
At this metod  values of fields are being assigned to VM properties. Further go bindings and so on.

>Do you have exception handling in place around all PLC interactions?

0.1.8 has **errorcode**, I used that.
0.8.1 doesn't have that. So right now is nothing.

>Does it always freeze after a few seconds (or what is the maximum time you've seen it running)?

seems the  time between  first "living number" read/change and the last  "living number" read/change is about 3 sec.
it may say it has constant interval before break.

>Also, not a technical issue, but could you describe how experienced you are as a C# programmer?

I am plc programmer, have some knowledge about XAML, C#, MVVM pattern. Consider me as a beginner.

bolt-b, 2021-03-08:
@mycroes 
simply amazed,
-launched app based on 0.1.8
-then launched app based on 0.8.1
this a marvel, both are working!
It looks like 0.1.8 something configured for 0.8.1. Perhaps Windows socket or so or configured plc . )
If app based 0.8.1 launch separately it freezes after ~3 sec.

mycroes, 2021-03-08:
> > Are you using a timer to read data?
> 
> Yes I am. It's System Timer.

What type of timer exactly? There's `System.Timers.Timer`, `System.Threading.Timer`, `System.Windows.Forms.Timer` and two others, they all start with `System` and end with `Timer` (but have very distinct behavior). Also, how many timer instances/events/callbacks are you using?

> > What thread is accessing the PLC instance?
> 
> I created a class library and connected it to solution. This class library has a class where I make a connection to plc.

That doesn't say anything about the thread that accesses the PLC instance.

> > Do you have multiple threads accessing the PLC instance?
> 
> This is quite big application, so my answer -yes. Yet while troubleshooting others are disabled.

The size of the application doesn't need to have anything to do with threading, so I'll just assume a _yes_ without considering the explanation. Do you use a synchronization primitive when accessing the PLC instance?

> > How do you update your UI?
> 
> An event subscribed to a timer, then it is being broadcasted to a method which is located in VM (used mvvm pattern).
> At this method values of fields are being assigned to VM properties. Further go bindings and so on.

Is the timer used to perform a read from the PLC or does it just copy the data from somewhere else?

> > Do you have exception handling in place around all PLC interactions?
> 
> 0.1.8 has **errorcode**, I used that.
> 0.8.1 doesn't have that. So right now is nothing.

Yes, 0.8.1 is less C-like and more C#-like in the sense that it'll throw exceptions when errors occur instead of returning error codes.

> > Does it always freeze after a few seconds (or what is the maximum time you've seen it running)?
> 
> seems the time between first "living number" read/change and the last "living number" read/change is about 3 sec.
> it may say it has constant interval before break.

Is there any relation between this interval and the timer intervals?

> > Also, not a technical issue, but could you describe how experienced you are as a C# programmer?
> 
> I am plc programmer, have some knowledge about XAML, C#, MVVM pattern. Consider me as a beginner.

Thanks for the honest answer. Many questions we get here are due to a lack of C# skills, a lot of the issues are not necessarily related to S7NetPlus, but to things like performing concurrent access to the PLC. Some of those things could perhaps be handled internally in the library, but the simple fact is they currently aren't. Other than that I do expect S7NetPlus to work just fine, I think we'd get much more issue reports if there was an actual issue with the current version. Currently 0.8.1 has been downloaded from NuGet over 2600 times, and there have been over 40.000 downloads since version 0.2.0 was released.

> @mycroes
> simply amazed,
> -launched app based on 0.1.8
> -then launched app based on 0.8.1
> this a marvel, both are working!
> It looks like 0.1.8 something configured for 0.8.1. Perhaps Windows socket or so or configured plc . )

The connections are transient, so are sockets. In other words, there's no such thing as a setup that persists across connections, let alone across application restarts. I guess this is just a coincidence, which logically provides more reason to assume the error is in the library usage. The fact that it doesn't happen with 0.1.8 is probably just as much a coincidence.

Also, is there any chance you could provide the code for your application? That might make it a lot easier to identify the issue...

TechByte-300, 2021-03-08:
I am sorry, let me take the opportunity since this branch looks alive.
May I ask something?
Does the lattest version keep the older way for writing bits?
And it must look as follows?

`sampleplc.Write("DB33.DBX0.1", 1);`

bolt-b, 2021-03-08:
@mycroes,
>What type of timer exactly? There's System.Timers.Timer, System.Threading.Timer, System.Windows.Forms.Timer and two others, they all start with System and end with Timer (but have very distinct behavior). Also, how many timer instances/events/callbacks are you using?

 
```
private readonly System.Timers.Timer _timer;
 _timer = new System.Timers.Timer(); 
 _timer.Interval = 250;          
_timer.Elapsed += OnTimerElapsed

if plc.IsConnected
_timer.Start();
```
 >Do you use a synchronization primitive when accessing the PLC instance?

I can't answer because I don't know what you mean by _synchronization primitive_ .

>Is there any relation between this interval and the timer intervals?

That's hard to say.

> I guess this is just a coincidence, which logically provides more reason to assume the error is in the library usage. 

Nobody disputes, so I wrote at header "Can't migrate... please help."
I believe that since 0.1.8 the lib has been changed somehow, so methods and conception I used while building of app don't fit anymore.
So I should do something with this.
Again, you may not believe in that, but app based on 0.8.1 is running without any freezes and breaks when app based on 0.1.8 is also been launched previously. Already tried a few times.

>Also, is there any chance you could provide the code for your application? That might make it a lot easier to identify the issue...
We are not allowed providing entire projects . Although I will try to replicate same issue on the sample, then upload.

Hope all my efforts gonna turn out with real help. )

mycroes, 2021-03-08:
> ```
> private readonly System.Timers.Timer _timer;
>  _timer = new System.Timers.Timer();
>  _timer.Interval = 250;
> _timer.Elapsed += OnTimerElapsed
>
> if plc.IsConnected
> _timer.Start();
> ```

Do you have only this single timer instance? Or do you create this per DB you're reading or something similar?
 
> > Do you use a synchronization primitive when accessing the PLC instance?
> 
> I can't answer because I don't know what you mean by _synchronization primitive_ .

I think [this](https://wiki.osdev.org/Synchronization_Primitives) could provide some insight, but I'll try to explain it really short. Let's assume you're reading data from the PLC and reading takes 1 second (I know it's a lot faster, but the scale is not important for the example). Now let's say you have two threads that need to read data. If you start reading DB1 at _T0_ the connection can only be used to wait for the response for DB1, which takes 1 second. If anywhere within that second the read for DB2 starts, you're creating concurrent access to the PLC class. When that happens there will be two threads contending to receive responses for their requests, which the library doesn't handle in it's current form. A synchronization primitive can be used to keep the read for DB2 waiting, until it has been given the clear when the DB1 read completes. There are many approaches to solving this and it's hard to decide for someone else what would be a good fit in their project, especially when you can't see the project.

On the other hand, if you only use a single timer (`System.Timers.Timer`, because this doesn't apply to all other timers) to read/write data you should normally not have to deal with concurrency, because `System.Timers.Timer` synchronizes it's execution.

> > Is there any relation between this interval and the timer intervals?
> 
> That's hard to say.

Yes, with 250ms interval that's indeed hard to say... You could perhaps apply logging to get more exact figures.

> Nobody disputes, so I wrote at header "Can't migrate... please help."
> I believe that since 0.1.8 the lib has been changed somehow, so methods and conception I used while building of app don't fit [...]

There are changes, but my best guess is that the failures are handled differently, so maybe there always were issues that went unnoticed because they had different impact. I can't say, but I'm very interested in how this turns out.

> Again, you may not believe in that, but app based on 0.8.1 is running without any freezes and breaks when app based on 0.1.8 is also been launched previously. Already tried a few times.

I have no reason to doubt you. It's strange that it works every time, that's definitely more than a coincidence. However assuming it's an issue with concurrency then switching the libraries might affect first execution of the application with regards to caching and optimization, I don't think we can get any better than this educated guess.... (Or perhaps we can when we figure out what's going on).

> > Also, is there any chance you could provide the code for your application? That might make it a lot easier to identify the issue...
> 
> We are not allowed providing entire projects .

I'd be willing to provide help (free of charge) in a way that suits your employer, for instance using remote assistance software like TeamViewer or by means of an NDA to prevent me from 'stealing' your code.

> Although I will try to replicate same issue on the sample, then upload.

I'm also interested in what happens when you perform the same actions in a simple command line application if you just do reads and a `Thread.Sleep(250)` to mimic the behavior of your UI. Let me know if you can replicate this in a sample that you can share though.

> Hope all my efforts gonna turn out with real help. )

I guess all people need to do here is to ask for help and provide information when requested to be helped. We are missing a proper sample so that people can take a look at that, it's something I want to add when I have time...

FalcoGoodbody, 2021-03-09:
Good morning @all

maybe I can provide a little testprogram which uses lib-version 0.8.1 and a timer to read and write a) multiple variables using a item list and to read b) a class within another DB inside the plc.
The plc is a S7-1515-2 PN which is in slot 1.
It is running since nearly 6:30 o´clock this morning without any exceptions.
Parallel to this little program my main logging is also running and reads far more values from this plc but with lib-version 0.7.xx

The test program is a console application.

```csharp
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using S7.Net;
using S7.Net.Types;
using libOptiplan;
using System.Timers;

namespace S7NetPlus_Testproject
{
    public class DB85
    {
        public int DINT { get; set; }
        public Int16 INT1 { get; set; }

        public Int16 INT2 { get; set; }
    }

    class Program
    {
        private static readonly string IPGlatt2 = "YOUR_IP";        
        private static readonly Plc myPLC = new Plc(CpuType.S71500, IPGlatt2, 0, 1);        
        private static byte[] byteArrayNonOptimized = new byte[100];
        private static System.DateTime start = new System.DateTime();
        private static System.DateTime end = new System.DateTime();
        private static TimeSpan durationRead = new TimeSpan();
        private static TimeSpan durationWrite = new TimeSpan();
        private static System.DateTime startOVerall = new System.DateTime();
        private static System.DateTime endOverall = new System.DateTime();
        private static TimeSpan durationOverall = new TimeSpan();

        // S7 ReadMultiple/WriteMultiple
        private static readonly List<DataItem> dataItemsRead = new List<DataItem>();
        private static readonly List<DataItem> dataItemsWrite = new List<DataItem>();

        static System.Timers.Timer ReadTimer;
        
        #region DataItems
        private static readonly DataItem varBit = new DataItem()
        {
            DataType = DataType.DataBlock,
            VarType = VarType.Bit,
            DB = 83,
            BitAdr = 0,
            Count = 1,
            StartByteAdr = 0,
            Value = new object()
        };

        private static readonly DataItem varByteArray = new DataItem()
        {
            DataType = DataType.DataBlock,
            VarType = VarType.Byte,
            DB = 83,
            BitAdr = 0,
            Count = 100,
            StartByteAdr = 0,
            Value = new object()
[... 252 lines cut ...]
            catch (PlcException e)
            {
                Console.WriteLine(e.ErrorCode.ToString());
            }                                  
        }

        private static void SetTimer()
        {
            // Create a timer with a two second interval.
            ReadTimer = new System.Timers.Timer(5000);
            // Hook up the Elapsed event for the timer. 
            ReadTimer.Elapsed += OnTimedEvent;
            ReadTimer.AutoReset = true;
            ReadTimer.Enabled = true;
        }

        private static void OnTimedEvent(Object source, ElapsedEventArgs e)
        {
            Console.WriteLine("The Elapsed event was raised at {0:HH:mm:ss.fff}",
                              e.SignalTime);
            Read();
        }
    }
}
```

bolt-b, 2021-03-09:
@mycroes, @FalcoGoodbody  and others, Hi

>On the other hand, if you only use a single timer (System.Timers.Timer, because this doesn't apply to all other timers) to read/write data you should normally not have to deal with concurrency, because System.Timers.Timer synchronizes it's execution.

I made **simplified** conception of a timer usage which has proven its reliabilty at industrial area since 0.1.8 appears here at "Releases" section. Weirdly enough 0.8.1 doesn't work in this context.

[image]

@FalcoGoodbody thanks for input especially for the real code I think it is worth to be studied!

mycroes, 2021-03-09:
@bolt-b This looks fine, I don't see any issues here as far as library usage is concerned. I'm wondering though how do you create the timer and how do you propagate your viewmodel events? The `System.Timers.Timer` documentation explicitly states you should set it's `SynchronizationObject` property to the form or control you're updating unless the timer is part of the UI, which it isn't, because it's in a class library. In that case the `SynchronizationObject` is probably `null`, which means the elapsed event will be run on threadpool threads and updating the UI from a threadpool thread causes issues. Some MVVM frameworks 'fix' this for you, others just provide the glue that you have to apply yourself to ensure the UI updates will end up on the UI thread. This could be a cause for UI hangs.

Lastly, do you have a try / catch around all of `OnTimerElapsed` code? I'd think an exception there would crash the application (by default), but still it could provide some insight.

scamille, 2021-03-09:
https://docs.microsoft.com/en-us/dotnet/api/system.windows.threading.dispatchertimer?view=net-5.0 could be an easy solution to ensure that the timer event runs on the UI thread.
Whether that is desirable when using synchronous S7NetPlus API is another question, it might make your GUI less responsive. But for that, async/await is probably a better solution anyway.

bolt-b, 2021-03-12:
@mycroes, hi
Continuing our conversation,
Beginning with 0.1.9  it is implemented COTP an TPKT classes.
Could you tell me in short what was a reason(s) of their implementation?

mycroes, 2021-03-12:
0.2.0 was my first release, which was 2 months later than the release introducing the COTP and TPKT classes, so I can only guess. Logically they were added as a means to apply separation of concerns and to isolate some protocol parts so they're not all over the codebase (as still is the case with many other stuff in S7NetPlus).

Perhaps you know that besides current maintainer of S7NetPlus I also have my own S7 connection library: [Sally7](https://github.com/mycroes/Sally7). If you only look at the front page of my project you'll see references to _COTP_ and _TPKT_ as well. I also have a TPKT class and a COTP namespace, however there's quite a difference in implementation between my library and S7NetPlus. I actually implemented Sally7 from the linked specs and blog posts, as such most of the protocol is actually pretty clear in Sally7 (vs the magic number riddle in S7NetPlus). Also, Sally7 currently only has out of the box support for reading and writing of DataBlock data (which is the only part I use professionally at this moment), for other datatypes one would have to inherit `IDataItem` and provide an implementation that matches the required data for the protocol (or ask me to implement it, as soon as someone asks I'll probably just add it).

Anyway, I don't think the reason behind the COTP and TPKT implementation will help us analyze your problem. Also, I don't think they're the cause of your issues, but it's hard to say with the limited information we have. If you feel like it, you could take Sally7 for a spin. I'm using that professionally, at current customers we're performing real time communication on over 1000 protocols using it, so I'd say that works pretty well. Doesn't solve your S7NetPlus issue, but I'm afraid that unless you provide something reproducible to us that it might stay just that: your issue.

bolt-b, 2021-03-13:
@mycroes,
Sure, you are right. Something reproducible is here [https://github.com/bolt-b/Examples](url).

How do I set/reset bits while reading? Could you make changes?
0.1.8 handled this easily, in comparison with 0.8.1 an usage of which hangs the App while writing.
-currently 0.8.1 is referenced to the project.
-you can easily switch over to 0.1.8. It is in project folder.

mycroes, 2021-03-13:
Hi @bolt-b,

> Sure, you are right. Something reproducible is here [https://github.com/bolt-b/Examples](url).
OK, so I took a look at your sample. First off, reading seems to work, the UI doesn't hang and I get values in the UI. However, I'm amazed it works at all, since you're not doing the UI updates from the UI thread. It might be MVVMLight is handling that for you (although my quick peek at it didn't seem to indicate that it does), but if it doesn't then this alone could be reason enough for the UI to hang.

> How do I set/reset bits while reading? Could you make changes?
No, what happens in my case is the following:
[image]
In this screenshot you can see the expected length of data to read is way too long, thus it will wait indefinitely on more data to read. This is because you're accessing the PLC object from multiple threads, as I was guessing about a week ago. The current implementation is fine, it's not thread-safe, but neither was the previous implementation. I'm not sure why the behavior differs (to this extent) between 0.1.8 and 0.8.1, but your sample is making improper use (concurrent access in this case) either way.

I also noticed something odd in the bit writing code when the value is an `int`, the code there just doesn't seem to add up logically. My suggestion would be to write `true` and `false` if you want to write bits instead of `int 0` or `int 1`, because writing an int for a boolean value doesn't make any sense (and neither does the code that is currently handling `int` value in `WriteBit`, as explained above).

Long story short, if you make sure you don't concurrently access the PLC, it'll probably all be fine. I'm thinking about a way that will permit concurrent requests, which would probably also permit concurrent access to the PLC object, however you shouldn't necessarily expect that on short term.

mycroes, 2021-03-15:
OK, so there's 2 things that made me reconsider concurrent requests (which would also handle concurrency in general):
1. Issues like this one, it's semi-hard to correctly use the library to begin with even though the fix doesn't need to be large
2. Performance

As for the first point, it's pretty obvious, it would make usage of the library easier if it could handle concurrency internally. The second point is a different story. The past few days I've worked on implementing concurrent requests for Sally7. Today I performed my first tests and exclusive vs concurrent requests made only a minor difference, concurrent requests were something like 3% faster. However, that was in a lab environment, i.e. the PLC and PC were probably (I was testing remotely, so I'm not sure) separated by one or two switches only. Also, the PLC only had a single datablock available, with a single variable. I couldn't change any of that, so I just put the same variable in a read request multiple times (and also tried with the single variable only).

So then I thought, let's test this at a customer site, with a server room, large network etc. I was reading a single `short[10]` array and there the concurrent reads were over twice as fast! Probably this also was a faster PLC, because the total time was reduced as well, but still this is a huge improvement.

Long story short, I figured how to do concurrent requests with Sally7, which works pretty well. I still have to cleanup the code and verify some things, but from initial testing it's looking really good. Now the thing is that the API surface of Sally7 is waaay smaller than S7NetPlus. Also, Sally7 only has an async api (because you know, network communication is inherently async), which is well suited for handling concurrency (because you can await available slots, freeing up threads that would otherwise be block by a sync operation). All in all I'm not sure if there's any better way than implementing all of this async in S7NetPlus as well, with perhaps a wrapper that exposes the sync methods using `.GetAwaiter().GetResult()`, but at least that's still an option.

I'm not sure yet when (or honestly, *if*, I'm still very time-constrained) I can add this to S7NetPlus, but at least if I want to take a stab at it the plan's already laid out.

scamille, 2021-03-16:
Btw I am using https://github.com/StephenCleary/AsyncEx for async locking of S7NetPlus. Works like a charm, and very easy to use with using/IDisposable.

bolt-b, 2021-03-16:
@scamille, 
it's getting so that I must dip my App in _async_ entirely. 
The other day came to the code as follows:

Seems damn freezings stopped. It needs to be tested comprehensively, though,
However, the _locker_ and _await_ are not compatible. The C# devs  say "to avoid" so called _deadlocks_.  
So it is impossible to use, say, _WriteClassAsync_ in a pair with _locker_.

**This works like a charm, but not _async_ at its end.**
```
 public async Task EmptyClassAsync(TransferToPlc transferToPlc)
 {      
            await Task.Run(() =>
            
               lock (_locker)
               {

                  plc.WriteClass(transferToPlc, 54, 0);

                }

            });
 }
```
Your tip leads to _AsyncLock_ that allows combination of _locker_ and _await_. Am I correct? )

**Your advice that also working like a charm, so to speak _full async_.** 
```
 public async Task EmptyClassAsync(TransferToPlc transferToPlc)
 {
            using (await _mutex.LockAsync())
            {
                await plc.WriteClassAsync(transferToPlc, 54, 0);

            }
}
```
So far I can't say which one is better as it is still being tested.
Generally speaking thanks for real help.

mycroes, 2021-03-16:
@bolt-b The latter will work fine, however you'd be better off moving it into a wrapper for the PLC instead of applying it anywhere you use the PLC (which simply put is applying DRY to your codebase). I had some ideas to change how communication in S7NetPlus currently works, I might take a stab at changing that this evening. In that case it'd also be easy to apply a synchronization primitive from inside the PLC class without much effort, which could solve all your problems entirely.

bolt-b, 2021-03-16:
@mycroes, 

**concerning teerminology:** concurrent reads/reequeests - you mean when two or more DBs are being read? Or constant reading of many DBs and occasional writing while reading?

> As for the first point, it's pretty obvious, it would make usage of the library easier if it could handle concurrency internally. 

ocassionally I look through all issues here, closed and opened. People use this lib differently. Some people perform a single reading once an hour and that all, some people trying to build some sort of SCADA or alternative of WInCC based on this lib. For the latter the concurrency is a panaсea. As there's a need in realtime values reading (as much real as it allows Windows)  with simultaneous writing.

> So then I thought, let's test this at a customer site, with a server room, large network etc. I was reading a single short[10] array and there the concurrent reads were over twice as fast! Probably [...]

I believe that  1511 and  1515(twice powerful) having identical DB _with a single short[10] array_ will produce same performance in terms of data exchange as both basically do(evaluate) nothing. 

 All in all I'm not sure if there's any better way than implementing all of this async in S7NetPlus as well, with perhaps a wrapper that exposes the sync methods using .GetAwaiter().GetResult(), but at least that's still an option.

mycroes, 2021-03-16:
> @mycroes hi,
> 
> **concerning terminology:** concurrent reads/requests - you mean when two or more DBs are being read simultaneously? Or constant reading of many DBs and occasional writing while reading?

In case of concurrent requests multiple requests can be send before the first requests has been responded to. Siemens PLC's actually support this, during connection negotiation a maximum number of concurrent requests can be negotiated, although I've never seen it higher than 3. What S7NetPlus and Sally7 traditionally have been doing (or expecting, because the libraries don't enforce it) is sending a single request, waiting for the response, then sending the next request. The advantage of concurrent requests is that while the PLC is still processing request A you can already send request B and C so that when the PLC is done it can immediately process the next request. And actually it seems like the PLC might be able to handle requests concurrently itself as well, because sometimes I get out of order responses (so sometimes the response for C arrives before the response of B).

> > As for the first point, it's pretty obvious, it would make usage of the library easier if it could handle concurrency internally.
> 
> ocassionally I look through all issues here, closed and opened. People use this lib differently. Some people perform a single reading once an hour and that's all, some people trying to build some sort of SCADA or alternative of WInCC based on this lib. For the latter the concurrency is a panaсea. As there's a need in realtime values reading (as much real as it allows Windows) with simultaneous writing.

Well I've done without concurrent requests for years (originally with Simatic NET / Sapi S7 we did do concurrent requests, but performance was still below par, although in part our own fault) and for long I've doubted the necessity or benefits, because in my lab testing it didn't offer much. And I guess I actually fall into your _SCADA or alternative of WinCC_ group (current customer example: 1 process reading 1551 variables and writing 1151 variables).

> > So then I thought, let's test this at a customer site, with a server room, large network etc. I was reading a single short[10] array and there the concurrent reads were over twice as fast! Probably this also was a faster PLC, because the total time was reduced as well, but still this is a huge improvement.
> 
> As example: I believe that 1511 and 1515(twice powerful) having identical DB _with a single short[10] array_ will produce same performance in terms of data exchange as both basically do(evaluate) nothing and have same ethernet controller onboard.

Is this guessing/assuming? I'm not sure about the performance between different PLC's, I'm a C# developer, I'm not involved with what our PLC department does and often I don't even see the PLC's when I'm onsite. I do know there's a setting that lets you choose the amount of processing time allowed for network requests. I'm not sure how that impacts performance either. The only thing I do know is that at above mentioned customer I could do 500 concurrent reads in 448ms, while exclusive reads took 1062ms (of exact same data).

bolt-b, 2021-03-18:
@mycroes hi,

> As example: I believe that 1511 and 1515(twice powerful) having identical DB with a single short[10] array will produce same performance in terms of data exchange as both basically do(evaluate) [...]

This is simple assuming based on observations.
There are few Siemens guys who might tell you how this actually behaves.
Because those guys who writes you oficial answers on your requests are mostly moonlighting students, say, from Bangalore.
Those guys who develops for Siemens at this area are simple not available for mere mortals.

>So far I can't say which one is better as it is still being tested.
Generally speaking thanks for real help.
>> @bolt-b, The latter will work fine.

I don't agree the latter simply glitches in my case,

I backed to this code, it is not stellar, but this works.

 ```
public async Task EmptyClassAsync(TransferToPlc transferToPlc)
 {      
            await Task.Run(() =>
            
               lock (_locker)
               {

                  plc.WriteClass(transferToPlc, 54, 0);

                }

            });
 }
```
Despite of the _AsyncLock_ allows to use library's  _Async_ methods with _lock_ statement,  it glitches.
As example: When I press button it doesn't affect anything immediatly, I go, say, take a cup of cofee, come back. Suprise! It affected the system. )

mycroes, 2021-03-18:
> I don't agree the latter simply glitches in my case,
>
> I backed to this code, it is not stellar, but this works.
>
> ```
> public async Task EmptyClassAsync(TransferToPlc transferToPlc)
> { [...]

Are you locking on the same object instance everywhere (for both reads and writes)? I can imagine all of this breaks when there's no synchronization at all, but it simply doesn't make sense no matter whether the async code is called using async locking primitives or the sync code is called using the `lock` statement. I must admit I don't actually use S7NetPlus, so I would probably be the last one to notice any issues in this area. I'll see if I can create some unit tests for this as well.

bolt-b, 2021-03-18:
>Are you locking on the same object instance everywhere (for both reads and writes)?
yes, I am.

mycroes, 2021-03-18:
Could you perhaps update your sample with the `AsyncLock`? I can take another look at that...

bolt-b, 2021-03-18:
>Could you perhaps update your sample with the AsyncLock? I can take another look at that...
made another upload and something deeesribed at issues section.

mycroes, 2021-03-26:
While I couldn't reproduce your issue using your sample yet, I've been hard at work implementing concurrent requests in Sally7. I intend to migrate that solution to S7NetPlus as well, might have some time to do that this weekend.

bolt-b, 2021-03-27:
@mycroes, answered you at rep.

mycroes, 2021-06-02:
Hi @bolt-b, it's taken ages, but I've just released v0.12.0, which adds synchronization on PLC calls. I'd love to hear if that solves your issues. In short, all locking etc. can now be removed, it shouldn't matter if you're using sync or async calls (though in all fairness, all sync calls will be executed sync while the internal queueing is using async, so it still applies `.GetAwaiter().GetResult()` to block), my recommendation is to use async though since that's what networking is anyway... Let me know how the new release works for you.

Regards,
Michael

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes