S7NetPlus/s7netplus #380: Using ReadMultipleVars with more than 20 data items throws an index out of range exception

thombrink, 2021-03-17:
When using more than 20 'DataItem' objects in a 'ReadMultipleVarsAsync' request, it throws the following exception:

`S7.Net.PlcException: Index was outside the bounds of the array. ---> System.IndexOutOfRangeException: Index was outside the bounds of the array. at S7.Net.Plc.ParseDataIntoDataItems(Byte[] s7data, List`1 dataItems)`

All 'DataItems' were of type 'VarType.Bit'.

Is this a known limitation? I couldn't find anyfhing about it in the wiki.

scamille, 2021-03-17:
Do you have multiple concurrent requests happening to the same PLC instance?

The exception message indicates that something about the data received is broken / does not match with what is to be expected, and concurrent PLC access is one of the likeliest reasons for that.

thombrink, 2021-03-17:
It's a single unittest (to a test server), so there are no other instances. If I lower the amount of DataItems to 19, it instantly works like expected.

scamille, 2021-03-17:
Hmm yeah after studying the code some more, I think the Size Limit check in https://github.com/S7NetPlus/s7netplus/blob/051091919fa80b03989f4375e364ebc4b02aabe7/S7.Net/PLC.cs#L185 is underestimating the amount of response data required.

=> So yes it is a limitation of the S7 protocol. S7NetPlus is just not pro-actively doing the correct limit checks to warn you against that.

thombrink, 2021-03-17:
Ah oke I see. Still wondering tho: the PDU size is set to a fixed number (240), because the it seems to choke on sizes above 256. The raw network stream returns 140 bytes in total, which is well below the limit. That doesn't seem richt?

Besides that: isn't there a way to notify the plc that we have read the buffer and are ready to read the next one?

scamille, 2021-03-17:
Or maybe https://github.com/S7NetPlus/s7netplus/blob/051091919fa80b03989f4375e364ebc4b02aabe7/S7.Net/PLCHelpers.cs#L243 is just wrong. I don't see why Byte and Bit should get extra padding even on even addresses.

At least it does not match the size calculation in the link I posted earlier.

scamille, 2021-03-17:
What Data Items are you reading exactly? 20 x 1 Single Bit per Data Item?

scamille, 2021-03-17:
```
        /// <summary>
        /// https://github.com/S7NetPlus/s7netplus/issues/380
        /// </summary>
        [TestMethod]
        public async Task Test_ManyDataitems()
        {
            Assert.IsTrue(plc.IsConnected, "Before executing this test, the plc must be connected. Check constructor.");

            var dataitems = Enumerable.Range(0, 22)
                .Select(i => new DataItem
            {
                VarType = VarType.Bit,
                StartByteAdr = i,
                Count = 1,
                DB = 1,
                DataType = DataType.DataBlock
            }).ToList();
            await plc.ReadMultipleVarsAsync(dataitems);
        }
```
Gives an error. I'll take a closer look.

scamille, 2021-03-17:
@mycroes Do you have an idea why this might happen? The test I posted makes it easy to replicate.

For me it looks like the PLC is just not sending back enough data, with no apparent reason. Parsing of the first 20 data items works correctly.
You might know more if there is some special S7 protocol limitation or something causing that.

thombrink, 2021-03-18:
I ran the sample you posted above against the shally7 package which @mycroes wrote and got the following exception: `System.Exception: 'Expected ItemCount 22, received 20.'`. So I guess it's a limitation of the s7 protocol.

thombrink, 2021-03-18:
I dug a little deeper and found out that the sharp7 package has a S7Client class which contains the following property: `'public static readonly int MaxVars = 20;'` I believe that says enough.

scamille, 2021-03-18:
That is definitely a pretty strong hint at 20 being the limit :-)
What S7 PLC Type are you testing this on? If this also happens on a S7-1500, I assume this is some limit set by the protocol / server itself. If this only applies to older PLC types, it might make sense to find out the limit on the newer types as well.

thombrink, 2021-03-18:
I'm not sure wich type, because I'm using the test server.

For now I have made the following simple workarround to request more than 20 variables:

```C#
for (var i = 0; i < variables.Length; i += 20)
    await _Plc!.ReadMultipleVarsAsync(requestDataItems.Skip(i).Take(20).ToList());
```

scamille, 2021-03-18:
If you have access to some Chunk(int chunkSize) Linq function, you could write it like this:
```
foreach(var chunk in requestDataItems.Chunk(20))
    await _Plc!.ReadMultipleVarsAsync(chunk.ToList());
```

Anyway, I'll prepare a fix to give a exception with proper Message directly in AssertPduSizeForRead.

I can also imagine enhancing ReadMultipleVarsAsync by automagically taking care of splitting up the requested work load so that the end user does not have to worry at all about it. Maybe just divide the number of data items by 2 every time the pdu size / number of items is over the limit. Repeat until things work out.
But someone would have to be willing to do the work for that :-)

thombrink, 2021-03-18:
Chunk would indeed be a nice solution. When implemented correct, it should cause almost no performance impact. The dataitems argument however should not be restricted to lists for that matter.

I'm not sure what advantage we would get by dividing the items?

Before I implement a more robust solution, I would like to know what @mycroes thinks about this problem in general.

scamille, 2021-03-18:
Well besides the 20 items limit, there is also the PDU Size limit restriction based on the actual data amount you are trying to read.

I guess more appropriate would be to iterate through the list of data items until you hit the ceiling (either because of size or 20 data items), and read the items you so far have accumulated.

Repeat that for the tail until done.

scamille, 2021-06-05:
@thombrink Can you say what specification your "test server" had when you stumbled upon this problem? Was that a real PLC, or just another Snap7 instance?

mycroes, 2021-06-21:
I'm definitely sure Snap7 cuts off at 19 items, while it will actually send back a response as if it had succeeded. Apparently we're writing over 30 (I believe 36, not exactly sure) items to an S7 1500 using Sally7, that shouldn't make any difference from S7NetPlus. I'd say there's room for improvement in the parsing code, but this issue is invalid. Closing, feel free to reopen.

closed 2021-06-21 by mycroes