S7NetPlus/s7netplus #384: ReadBytes Exception

zzzdddyyy, 2021-04-13:
Function:
       I want read 1 byte from DB1,actually,DB1.0  is 1,0,0,0,0,0,0,0; The return value of this method should be [1],just 1 bytes,but somtimes i receive 15 bytes.
code:
        `byte[] rePLC_Status = MyPLC.ReadBytes(DataType.DataBlock, 1, 0, 1);`
Exception:
        S7.Net.PlcException:“Received 15 bytes: '32-03-00-00-00-07-00-02-00-01-00-00-05-01-FF', expected 19 bytes.”

mycroes, 2021-06-01:
Hi @zzzdddyyy,

The 15 bytes received are part of the total message, which includes a header before the actual response. If it's only 15 bytes (instead of the expected 19). What's strange though is that the last byte should be the result code given by the PLC, and `FF` is the code for success. Is there any chance you're having concurrent access (multiple threads) to the PLC instance? That's currently not supported and does tend to lead to similar issues. #394 does address this issue though.

mycroes, 2021-06-02:
Hi @zzzdddyyy, please retest this with v0.12.0.

zzzdddyyy, 2021-06-07:
Thanks for your response.Yes,I used a timer loop to read DB variables.I will try to use the latest version

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes

Jason-Jelks, 2021-10-09:
When I ran into a possibly similar problem recently,  I found that instantiating an instance with the overload PLC(string ip, TsapPair tsapPair) explicitly from my code reverts to S7-300, but if when I changed to the standard public Plc(CpuType cpu, string ip, int port, Int16 rack, Int16 slot), the issue resolved. If that doesn't help at all, it may help to understand what the structure (DB, Array, Input, Output...) you are attempting to read. 
We have S7-1212 and S7-1515F-2 PN Safety in house that we test our S7netPlus changes with.