S7NetPlus/s7netplus #387: Ask for a Concurrent Version

mabc666, 2021-04-17:
My app has a read task that  circularly read a lot of db block and i wannna send some values into db at same time. I guess that's going to be a problem. Is it right?

scamille, 2021-04-17:
Yes the Plc object can only be used by a single thread at a time. So you need to queue your write command somewhere in between your reads, but you can't do it concurrently.

You could try using 2 separate PLC objects, butthat might not always work, depending on your PLC.

mabc666, 2021-04-17:
thanks for your answer .
I really like the library your wrote. I used to wrap the snap7 to support concurrent, i think it's a useful thing that we can use this library like idiots.

mycroes, 2021-06-01:
@mabc666 #394 should solve this. If you want you can give the code from the PR a spin to see if it works.

mycroes, 2021-06-02:
@mabc666, I'd say go have fun with v0.12.0, it should now work with multithreaded access just as well.

closed 2021-06-02 by mycroes