S7NetPlus/s7netplus #392: While Writing Integer Value;

jaydeep6720, 2021-05-27:
//Below code is to write VW2 address
        private void button1_Click_1(object sender, EventArgs e)
        {
            int VW0 = int.Parse(txbVW0.Text);
            s7200.Write("DB1.DBW0", VW0);
        }

Above code writes Value of VW2 in S7 200 SMART PLC.
Remaining everything is OK.

mycroes, 2021-05-31:
@jaydeep6720 I'm not entirely sure what your question is, but one minor detail that I'm not sure about the implementation in S7NetPlus is that you're writing an `int`, which is 4 bytes, while a word is 2 bytes. I'd say the `Write` method should take care of writing only the addressed variable (so 2 words only), so let me know if that's what you're talking about...

jaydeep6720, 2021-06-01:
Dear sir
I want say that when i write VW0 absolute address from S7 net plus, in PLC it is writing VW2 instead of VW0,

mycroes, 2021-06-01:
Dear @jaydeep6720,

As I guessed, this is related to the fact that the value is written, regardless of the specified type in the address. Small explanation, `INT` in Siemens is a 16 bit (2 bytes) signed integer, `int` in C# is a 32 bit (4 bytes) signed integer. As a result, your data ends up in byte 4 if it's less than 256, in byte 3 and 4 if it's less than 65536 (as a result of how the data is stored in the PLC). However, if you use the C# type `short` you're having the equivalent of the Siemens `INT`, so the following code should work:

```csharp
// Below code is to write VW2 address
private void button1_Click_1(object sender, EventArgs e)
{
    short VW0 = short.Parse(txbVW0.Text);
    s7200.Write("DB1.DBW0", VW0);
}
```

Regards,
Michael

closed 2021-06-01 by mycroes