S7NetPlus/s7netplus #393: Multiple call ways

jaydeep6720, 2021-05-27:
Hello all
Below mentioned function is working fine. but the point is i need to call this same function for more then 100 times.
is the anyway in which i can write this as a function or class and useit for 100 variable ??

var M22 = ROT.Read("M22.0");
            bool M22r = Convert.ToBoolean(M22);
            if (M22r==true)
            {
                btnDEXP.BackColor = Color.ForestGreen;
                string btnM22on = "ON";
                btnDEXP.Text = btnM22on;
            }
            else
            {
                btnDEXP.BackColor = Color.Red;
                string btnM22off = "OFF";
                btnDEXP.Text = btnM22off;
            }

mycroes, 2021-05-31:
@jaydeep6720 I'm not sure what you're asking. What's stopping you from creating a method that does what you've written above?

jaydeep6720, 2021-06-03:
found the way. :)
forgot to update here.

closed 2021-06-03 by jaydeep6720