S7NetPlus/s7netplus #399: Default connection timeout

mycroes, 2021-06-06:
Currently unset, could use a default value. Thinking about 10 seconds default, anyone could override if that's not enough. Users of the library are more likely to wait for the 10s timeout than a 1m timeout, so I'd rather set this low by default. Was also shortly brought up in #295.

@scamille @jakob-ledermann Any opinions on this?

scamille, 2021-06-07:
I am strongly in favor of setting a default timeout. We are talking about read/write timeout and not connect timeout, right?

While historically there are often larger timeouts used (eg 30s command & connection timeout in SQL), nowadays that is quite a long time. 10s sounds like a sensible choice. For a modern S7 1500 CPU this could likely be a lot lower, but it might break slower setups still likely encountered in real systems.

mycroes, 2021-06-07:
Well the send/receive timeouts also affect connect. I'm not sure what affects the `TcpClient.Open()` call, but in all honesty you can roundtrip the world in less than 100ms. That doesn't mean I'm in favor of a 500ms timeout, but I think 10s is fair as long as users can change this.

jakob-ledermann, 2021-06-07:
I think 10 seconds is reasonable, as the main goal is not to react to problems as fast as possible, but to make sure users do not wonder why their software no longer talks to the PLC. As far as I know there is no socket option on windows which affects the initial TcpClient.Open() timeouts, so that should remain unaffected.
And yes IMHO users must be able to change this. A receive timeout of only a few milliseconds may easily trigger even on a heavily used TCP connection.

PR #402 (2021-06-07): Default ReadTimeout and WriteTimeout to 10 seconds
Fixes #399
closed 2021-06-07 by mycroes