S7NetPlus/s7netplus #403: Help! I ran into some problems

wanqi121, 2021-06-21:
I Read the data of DB1  
ushort result = (ushort)SIMATIC_PLC.Read("DB1.DBW0");
Return error "Received 12 bytes: '32-02-00-00-00-00-00-00-00-00-81-04', expected 20 bytes."
Can someone tell me what is going on?

mycroes, 2021-06-21:
Hi @wanqi121,

Probably you forgot to enable *Permit access with PUT/GET communication from remote partner (PLC, HMI, OPC, ...)* in TIA portal. Please report back if that was indeed the issue.

Regards,
Michael

scamille, 2021-06-22:
Is there a way we can parse this response to assert that missing put/get communication is the problem, and throw a better exception?

With the first 4 bytes being TPKT header, I have a hard time parsing the TPDU. Its dynamic header length is 0 which we don't handle well but that could be changed. Problem is that I can't even fit those 8 bytes into the TPDU header without any dynamic length, and that means I can't make much of the last 2 bytes which would have non-zero values in this example.

mycroes, 2021-06-22:
@scamille I'm not sure about all the zeroes, but 0x8104 _probably_ is the error that PUT/GET is not enabled. Also, I'm not sure if first 4 bytes here are actually TPKT, or if this is just the TPKT contents.

scamille, 2021-06-22:
Oh right of course all of that data is just S7 payload. Too early in the morning.

I think we generally under-parse the S7 header of the read response in AssertReadResponse

http://gmiru.com/article/s7comm/ neatly explains the data here a bit:
The second byte being 0x2 identifies this as a ACK message, with 0x81 being the error class and 0x4 the error code.

irongiants, 2021-06-30:
Hi guys,I met the same question.The reason is not the configuration . Finally I resolved it by updating the Siemens TIA_Portal_STEP_7_Pro software from v14 to v15.

bartekbb, 2021-09-09:
Hello, I got same problem
My PLC is S7-1200 (CPU 1211C DC/DC/DC), access with PUT/GET is enabled, firmware v4.2, using TIA V16, DB2 has unchecked optimised access
trying the simplest data read method:
data1.Text = ((ushort)plc.Read("DB2.DBW4")).ToString();

[image]