S7NetPlus/s7netplus #404: basler camera damaged

desmondlee, 2021-06-24:
Hi there,  when I use this lib to connect plc s7-1200,   

the basler camera will be damaged (the grabbing state of basler camera will be changed from "true" to "false", and it can't be restart again).

this almost killed me :(

mabc666, 2021-06-25:
你的PLC和相机是通过不同网口连接的吗？

mycroes, 2021-06-25:
@desmondlee S7NetPlus doesn't do any magic, perhaps this was just a coincidence. Did you actually write anything at all, or just connected to the plc?

desmondlee, 2021-06-28:
> @desmondlee S7NetPlus doesn't do any magic, perhaps this was just a coincidence. Did you actually write anything at all, or just connected to the plc?

just connected to the plc.

mycroes, 2021-06-28:
@desmondlee I'm confident this has not relation to S7NetPlus in any way. I don't know how you connect your camera to the PLC, but it already seems like an issue if PLC could contribute to such issues at all.

We also use the S7 1200 with camera's, but we use Allied Vision camera's, not Basler. I know that depending on the mode of the camera it's impossible to grab an image by request, for instance if it's set to capture on external trigger.

Anyway, closing this issue because it's not S7NetPlus related, feel free to report back what the actual issue was.

closed 2021-06-28 by mycroes