S7NetPlus/s7netplus #406: Connect via/over NAT

RegyRegy, 2021-07-12:
Hello,
is there an option to connect to the S7 behind the router over NAT ?
I think it should be possible by forwarding ports, but can we set connection port with S7.net ?

Regy

scamille, 2021-07-12:
You can set the port with this constructor: https://github.com/S7NetPlus/s7netplus/blob/b475aee2e768f709e6b8f01084b176e53d43cd8c/S7.Net/PLC.cs#L144 when creating the PLC object.

Using a different port than 102 for S7NetPlus works fine, assuming your PLC can be reached through it.

RegyRegy, 2021-07-12:
> public Plc(CpuType cpu, string ip, int port, Int16 rack, Int16 slot)

What a shame I didn't notice that! 
Thank you @scamille ! It't time to test it and forward ports.

I will give a feedback later!