S7NetPlus/s7netplus #407: How do I handle with exceptions occuring while...

Wobeam, 2021-08-05:
Hello,

A question: How do I handle cases when:
1) PLC is not available.
2) Connection is broken.

I mean that  you removed IsConnected, IsAvailable  properties.

Could I look at example?
Thanks!

FalcoGoodbody, 2021-08-10:
Hello @Wobeam ,

in my current project I integrated a list of strings containing the ip-adressess of all my plcs.
I then give this list to a method which tries to ping all these ip-adressess and writes the results in a list of PingReply.
Afterwards I check this list if the status for each ping is "success". Then I try to open a connection to this IP and start reading/writing my data.

For your second question you could use a variable which will be incremented from one site (plc or your program).
The other side can then check the new state with the old state. If it is the same, then the connection is maybe broken.

Wobeam, 2021-08-10:
Hi Falco, probably you saved me! :+1:

scamille, 2021-08-20:
The core of your code should look something like this:

```
if (!plc.IsConnected)
{
  plc.Connect();
}
plc.Read(...);
```

It can always happen that you get a exception when calling Plc.Read/Write functions. S7NetPlus should automatically reset the tcp socket if the connection is broken. The next time you execute your code, a new connection to the plc would be established.

You can not 100% reliably determine if a connection is still functional without actually reading/writing data, so the correct way is just to handle errors there in a robust way.