S7NetPlus/s7netplus #411: S7.Net.TPKTInvalidException: TPKT is incomplete / invalid

kaustas, 2021-08-18:
Hello. I'm facing an issue with connection to PLC.

There are two PLCs (S7-300) to connect with one application. Problem happens when the second instance of application is started: one of PLCs failed to connect with TPKT is incomplete / invalid error message. At the same time second PLC connects without problem. Doesn't matter at what PC application's copy starts on same or different.

I thought there could be some limit for connections amount, but I didnt find any information about 102 port limitations and ISO on TCP.
[image]

I have no idea why this happens.

scamille, 2021-08-20:
Do I understand this correctly: You have a application which connects to both PLCs. The first instance of your application works fine with both PLCs. Once you start a second instance of your application, only one of the two PLCs works in this second application instance. And I presume the first application instance works fine the whole time.

Is that correct?

kaustas, 2021-08-20:
Yes you understood this correctly.

scamille, 2021-08-21:
Ok. I don't know much about connection limits for S7 plcs, but I would assume that there is something restricting the number of connections on your second plc, at least more tightly compare to the other one.

S7NetPlus currently does not fully parse all error codes in the responses it gets, so the TPKT is incomplete / invalid message might in fact contain something stating explicitly that some limit has been reached.

rafasimple, 2021-09-22:
Hi i have the same problem, the TPKT is incomplete, , but i dont know why, whit CPU 317 with CP343-1 Siemens,the error appears usually and then dissapears.

Alkaidxx, 2022-01-26:
when i use the CP243-1 to connect s7-200 with s7.net(use plc.open) , this problem(the TPKT is incomplete) is also appened. has anyone solved it?

sebastianhofer, 2024-01-27:
Is there any update on this issue? I have the same problem with S7-300 CPU, with the latest S7.Net Version 0.20.0.

sebastianhofer, 2024-01-29:
> You can try earlier Version

We fixed it with restart of the CPU (STOP -> RUN). After that the same lines of code works fine.

superbug8848, 2024-03-10:
I have same problem , CPU is S7 416， when my program want to connect to PLC, get a exception with “TPKT is incomplete / invalid ”， I use S7.net 0.20.0

Ced-90, 2024-08-23:
Hi all,

I have this issue when I try to write a string, sometime on S71500 / S7.net 0.20.0 (but not every time and not with all PLC).
(I have the TPKT error only with S7String so far)

It fail if I use for this:
`PLC.Write(DataItem_Array[]);`

or this : 
`PLC.Write(DataItem);`

but the error disappear and the string is written if I use
`PLC.Write(DataType.DataBlock, DataItem.DB, DataItem.StartByteAdr, DataItem.Value); `