S7NetPlus/s7netplus #414: ReadClass sbyte, uint, long, ulong aren't read

usingnamespaces, 2021-09-30:
I'm using the Method ReadClass.

For the datatypes 
DWord, have to use int as c# type instead of uint.
SInt, have to use byte as c# type instead of sbyte.
UDInt, have to use int as c# type instead of uint.

It's working meantime. I have to cast the properties back to the desired type.
Is it possible to fix this.

The PLC datatypes LWord, LInt, ULInt (all 64bit) aren't working at all if I declare them as long or ulong.

Jason-Jelks, 2021-09-30:
Which version of S7Net+ are you currently using?

usingnamespaces, 2021-09-30:
0.13.0

scamille, 2021-10-01:
Sbyte might not be supported, but a quick glance a the code suggests that uint should work fine.

What happens if you read a DWord or UDInt variable into a C# uint property?

usingnamespaces, 2021-10-04:
Structure of my datablock.
[image]

Structure of the class.
[image]

When variable 4 is read the actual value in the PLC is 3. The value read is 50331648.
For variable _11 the actual value in the PLC is 10. The value read is 167772160.

here the results
[image]

scamille, 2021-10-04:
Ok thanks for the data. I think I found the problem at https://github.com/S7NetPlus/s7netplus/blob/b475aee2e768f709e6b8f01084b176e53d43cd8c/S7.Net/Types/Class.cs#L169

Will try to create a fix for it.

Jason-Jelks, 2021-10-04:
For what its worth @scamille, I didn't respond to this thread yet, but I have almost completed the revisions for 64-bit values which I should be committing today. Up to you if you want to do it also.

mycroes, 2021-10-04:
I'm sorry guys, I'm a bit short on time (as usual), but I'll gladly accept a PR in similar style as other types.