S7NetPlus/s7netplus #420: S7: How do I protect plc while usage this library.

bolt-b, 2021-12-02:
Hello everyone,
From documentation we all know that _plc security option_ must have **full acccess(no protection)** checked and Permit accerss with Put/Get must be also checked.

Thus we provide full acccess to everyone. That is bad.
So my question is, are there ways to ptotect plc from everyone, of course leaving access for application based on s7net library.

thanks

Jason-Jelks, 2021-12-02:
I just verified on an S7-1515F-2PN (Safety PLC) that you should be able to set at [Read Only], and still connect with the software. We are able to read and write to UDT's that are set with Put/Get permission.. so, I am not 100% positive what exactly [Read Only] means if we are able to write, but you could research Siemens Documentation to find more.

Jason-Jelks, 2021-12-02:
Per [Siemens Support Documentation](https://support.industry.siemens.com/cs/mdm/109741593?c=91472806539&lc=en-TW)
**Full access (no protection)** :  Allows full access without password protection.

**Read access** :  Allows HMI access, comparing Offline/Online code blocks, and all forms of PLC-to-PLC communications without password protection.Password is required for modifying (writing to) the CPU. Password is not required for changing the CPU mode (RUN/STOP).

**HMI access** : Allows HMI access and all forms of PLC-to-PLC communications without password protection.Password is required for reading the data in the CPU, for comparing Offline/Online code blocks, for modifying (writing to) the CPU, and for changing the CPU mode (RUN/STOP).

**No access (complete protection)** :  Allows no access without password protection.Password is required for HMI access, reading the data in the CPU, comparing Offline/Online code blocks, and for modifying (writing to) the CPU.

bolt-b, 2021-12-02:
Of course, the chapter about security levels has been read before.

I say more since I'm currently playing with different security levels of plc using my App and pure TIA Portal with not loaded project. As though an outsider tries to get access.

The level called "HMI", which still allows Put/Get access, but disables any Reading/Writing by means of TIA Portal. However, anyone can see the program structure (function block's and DB's tree and their names).

The most interesting level called "No access". It disables Put/Get access and don't allow seing program structure. it asks for password for HMI.

My question is how do I implement in my App auto prompting and answering with a password for HMI level?
it would be cool to have that.

scamille, 2021-12-02:
As far as I know, the S7 protocol just does not support any security and also requires this open access.

You could look for an alternative protocol like OPC UA on the S7 1500, or ask Siemens if they can provide you with a better alternative to that,

or you can live with the limitations of the S7 protocol and implement as much network/physical security as you can around it.

bolt-b, 2021-12-02:
> As far as I know, the S7 protocol just does not support any security and also requires this open access.

I don't know whether S7 protocol supports or not security, but wide known WinCC when it starts it prompts for a password in case when access to plc is limited by "HMI level". That's the point.

Jason-Jelks, 2021-12-02:
On that note, Siemens does support TCP/IP (as host) if you want to take the time to create the TCON in the PLC. Of course you will also have to write the TCP Client and parse the packets. You would need to investigate to see what level of security TCP requires and if you can possibly establish an SSL/TLS connection.  The primary difference is that someone cannot 'snoop' your PLC using TCP, and you do can use optimized objects.

mycroes, 2021-12-03:
Actually the S7 protocol does support password 'protection'. I'm not sure where I've read about it, but IIRC basically it does a simple encoding which can easily be decoded by snooping the network traffic. Also, I'm quite sure the PLC itself can't protect against packet source spoofing, so basically S7 protection is rendered moot by the bad implementation.

However, there is one reason it can have added value: with no protection the default connection TSAP's can be used to connect to the PLC, which permit any source address to connect (vs only configured source addresses if you define an S7 connection in TIA Portal).

I don't have any plans to add the password support, but I'm not saying I will never add it. It's an interesting and perhaps unique feature for open source S7 libraries, but I don't have any personal use for it, nor can I easily test it.

mycroes, 2021-12-03:
> On that note, Siemens does support TCP/IP (as host) if you want to take the time to create the TCON in the PLC. Of course you will also have to write the TCP Client and parse the packets.

The S7 protocol uses so-called ISO-over-TCP, which as the name implies, uses TCP.

> You would need to investigate to see what level of security TCP requires and if you can possibly establish an SSL/TLS connection.

TCP doesn't offer any security on its own. I'm not sure about SSL/TLS, you could of course implement that in software but I'm afraid the speed would be awful.

> The primary difference is that someone cannot 'snoop' your PLC using TCP,

Not sure where you're getting this from, but TCP doesn't offer any protection on snooping. Networking hardware might try to avoid this, but unless you actually apply something like IEEE802.1X (wired network security, supported by 1500 series PLC's) anyone with physical access can do whatever they want. And then there's still the risk of intrusion on the controlling PC, in that case the password probably isn't going to help either (because there's a fair chance it'll be stored in plain text, either as source or somewhere in the process of sending data to the PLC).

mycroes, 2021-12-03:
> You could look for an alternative protocol like OPC UA on the S7 1500, or ask Siemens if they can provide you with a better alternative to that,

I believe OPC UA uses client certificates to verify the connection, that's certainly more secure than the S7 password protection.

> or you can live with the limitations of the S7 protocol and implement as much network/physical security as you can around it.

This is probably the best advice given in this entire thread. If there's no physical security, it's not secure.

Jason-Jelks, 2021-12-06:
The statements you make about security are absolutely correct, and I was not stating that TCP was secure. The question was: is there an alternative. 

What I was eluding to with the TCON functions, is that these are alternative functions in Siemens, T-Send/T-SendC and T-Recv, which establish a TCP packet protocol. Those TCP communications are explicitly defined with data types and packet structure. I was not implying that they are secure outside of the fact that a hacker would have to know the signature to establish the connection. I know this is high level, but this is not the forum to delve into the details about how it functions and how to set it up.

Jason-Jelks, 2021-12-06:
Just for additional reference so you don't have to start looking from the beginning, here is a link to Siemens doc for TCP, Secure-TCP. I hope this helps.

[TCP Base Communications](https://cache.industry.siemens.com/dl/files/710/109747710/att_992591/v2/109747710_Tcp_BaseComm_V1_en.pdf)